AdnanHodzic/auto-cpufreq #15: psutil has no attribute 'sensors_battery'

Exorust, 2020-01-06:
Even after I fix errors due to the fact that certain distros don't have /usr/lib/systemd/system/ and instead have /lib/systemd/system/, I still encounter error with psutil.
[image]

Exorust, 2020-01-06:
Turns out ubuntu 16.04 ships with older version of psutil when you use, `sudo apt install python3 python3-distro python3-psutil python3-click -y`

AdnanHodzic, 2020-01-07:
Yes, that's correct, could you please try upgrading both using pip3 by running:

`sudo pip3 install --upgrade psutil`

And let me know if the problem still persits after this?

I'm also going to update README with this information

AdnanHodzic, 2020-01-07:
Since I also [updated README with this information](https://github.com/AdnanHodzic/auto-cpufreq#requirements-installation-for-debianubuntu-and-their-derivatives), I'm closing the story. Feel free to re-open it if the issue persists.

closed 2020-01-07 by AdnanHodzic