AdnanHodzic/auto-cpufreq #16: The daemon stops the moment I pull out the plug.

Exorust, 2020-01-06:
Everytime I pull out the plug or plug the laptop in, the daemon crashes.
[image]

AdnanHodzic, 2020-01-07:
As you suggested, I think problem is identical to: https://github.com/AdnanHodzic/auto-cpufreq/issues/21

As I suggested there, I think these problems could be related to using older package version for "psuti"l which were installed from apt repos:

`sudo pip3 show psutil | grep -i version`

Recommended version should be >= 5.5.1

Hence, could you please try upgrading both using pip3 by running:

`sudo pip3 install --upgrade psutil`

And let me know if the problem still persits after this?

AdnanHodzic, 2020-01-07:
Issue is duplicate of: https://github.com/AdnanHodzic/auto-cpufreq/issues/12

closed 2020-01-07 by AdnanHodzic