AdnanHodzic/auto-cpufreq #17: Error: Couldn't find any of the necessary scaling governors.

sandersantema, 2020-01-06:


AdnanHodzic, 2020-01-07:
When I was making this tool I was strictly targeting Intel, as I didn't have any AMD CPU's to test it. Either way thank you for updating me about this.

As a goal to extend this tool for AMD as well, could you please confirm that in AMD's case, "ondemand" acts like "powersave" for Intel?

And please correct me if I'm wrong, but it seems you have following governors: `ondemand performance schedutil`?

`sudo tlp-stat -p` shows it, but you can also verify by running: `sudo cat /sys/devices/system/cpu/cpu0/cpufreq/scaling_available_governors`

sandersantema, 2020-01-07:
> As a goal to extend this tool for AMD as well, could you please confirm that in AMD's case, "ondemand" acts like "powersave" for Intel?
 
What exactly do you mean by this? Do you want to know whether the governors are practically equal? I would need to look into it. I might be able to find something about that on that on the internet, but if would like me to test things on my hardware I would need to have some more info. 

> And please correct me if I'm wrong, but it seems you have following governors: `ondemand performance schedutil`?
> `sudo tlp-stat -p` shows it, but you can also verify by running: `sudo cat /sys/devices/system/cpu/cpu0/cpufreq/scaling_available_governors`

No, you're right but unfortunately it seems auto-cpufreq doesn't recognise them. `sudo cat /sys/devices/system/cpu/cpu0/cpufreq/scaling_available_governors` puts out the same thing.

AdnanHodzic, 2020-01-07:
> What exactly do you mean by this? Do you want to know whether the governors are practically equal? I would need to look into it. I might be able to find something about that on that on the internet, [...]

From my knowledge of AMD CPU's (which is limited), I think `ondemand` acts and does same thing `powersave` does on Intel CPU's. I asked just in case you knew this information, but since you don't it's fine don't sweat it.

I need one last thing from you, based on which I'd like to make experimental support for AMD, which you could help me by testing it on your hardware. We'll do this in "live" mode which won't make any permanent changes to your computer. So no harm will be done.

In `auto-cpufreq/scripts` directory, could you run:

`cpufreqctl --driver` and tell me what output do you get?

sandersantema, 2020-01-08:
Thank you for your interest in supporting AMD. `auto-cpufreq/scripts/cpufreqctl.sh --driver` outputs `acpi-cpufreq`. 

Assuming the AMD beta support will not lead to hardware damage I'd be willing to test it out :)

rur0, 2020-01-14:
Hi
I'm also trying to run this program on a Ryzen 7 3700U CPU and getting the same error.

it looks like I only have access to "performance" and "schedutil" governors.
```bash
$ cat /sys/devices/system/cpu/cpu0/cpufreq/scaling_available_governors                            
performance schedutil
```

to add, I am using kernel 5.4.6-2

Thanks

e4BdSBmUzHowFico5Ktn, 2020-04-02:
@rur0 that is true, anything older than AMD 4000 (which is yet to be launched) does only have 2 power states available in any OS. AMD 4000 series cpus [will have more power states](https://www.pcworld.com/article/3530289/ryzen-4000-cpus-amd-zen-2-laptops.html) available. 

However, I don`t think this is a reason not to add functionality for any <4000 family ryzen cpus.

q-codes, 2020-04-08:
@AdnanHodzic  Having this same error in _Intel(R) Core(TM)2 Duo CPU_ . 

`/sys/devices/system/cpu/cpu0/cpufreq/scaling_driver    = acpi-cpufreq
/sys/devices/system/cpu/cpu0/cpufreq/scaling_governor  = ondemand
/sys/devices/system/cpu/cpu1/cpufreq/scaling_available_governors = ondemand performance schedutil `

alkapwn3d, 2020-05-23:
cpu: Intel i3 M 370

getting the same error. I only have performance and schedutil governors as well.

would love to see what this does for my laptop performance

Red-Eyed, 2020-08-04:
@sandersantema , I could try to create branch for you that could solve your problem (just add switching from `performance` to `schedutil`).  
Will you able to test it?

PR #67 (2020-08-05): Refactroring and amd support
fixes #17

- Added get_avail_powersave() (and performance).
  This function returns the most energy(and performance) efficient governor.
  This should fix AMD issue when powersave is not available
  but "ondemand" or "schedutil" are present

- Cleaned up and refactored.
  Added more convenient util functions.
  Also made it more pythonic, reduced subprocess calls, made it more modular

- Added support for setting turbo boost for AMD and Intel (intel_pstate and cpufreq driver)

- Tested on Intel and AMD machines and it works fine
    - It at least doesn't  fail
     [... cut]
PR #73 (2020-08-06): K10temp
Fixes #17 
Make it more stable: if sensors not found: print "nan"
closed 2020-08-05 by AdnanHodzic