AdnanHodzic/auto-cpufreq #18: Traceback when launching

e4BdSBmUzHowFico5Ktn, 2020-01-06:
I receive this traceback when running `sudo python3 auto-cpufreq.py`

`Traceback (most recent call last):
  File "auto-cpufreq.py", line 382, in <module>
    @click.option("--daemon", is_flag=True, hidden=True)
  File "/usr/lib/python3/dist-packages/click/decorators.py", line 170, in decorator
    _param_memo(f, OptionClass(param_decls, **attrs))
  File "/usr/lib/python3/dist-packages/click/core.py", line 1460, in __init__
    Parameter.__init__(self, param_decls, type=type, **attrs)
TypeError: __init__() got an unexpected keyword argument 'hidden'
`

Exorust, 2020-01-07:
You need to update your click version. Please see issue #4 .

AdnanHodzic, 2020-01-07:
Yes, this is correct, please do what @Exorust suggested.

e4BdSBmUzHowFico5Ktn, 2020-01-07:
Thanks, that solved it, as was mentioned in #4 the instructions should indeed be updated, I`m pretty sure its possible to attach library versions to install commands. In conda its something like:
`conda install PACKAGE -vx.y`

closed 2020-01-07 by e4BdSBmUzHowFico5Ktn