AdnanHodzic/auto-cpufreq #23: Systemd service fails on boot but can be manually started.

dominicdoty, 2020-01-08:
Systemd service fails on boot, but runs successfully when started manually (sudo systemctl start auto-cpufreq.service). Suspect that the service needs a different "After" target. Will dig further on my own.

System:
Arch Linux x86_64 5.4.8-arch1-1
Intel i5-4200U
Asus Q501LA 1.0

sudo systemctl status auto-cpufreq.service immediately after boot:
```
● auto-cpufreq.service - auto-cpufreq - Automatic CPU speed & power optimizer for Linux
     Loaded: loaded (/usr/lib/systemd/system/auto-cpufreq.service; enabled; vendor preset: disabled)
     Active: failed (Result: exit-code) since Tue 2020-01-07 18:31:17 MST; 15min ago
    Process: 474 ExecStart=/usr/bin/python3 /usr/bin/auto-cpufreq --daemon (code=exited, status=1/FAILURE)
   Main PID: 474 (code=exited, status=1/FAILURE)

Jan 07 18:31:17 orpheus systemd[1]: Started auto-cpufreq - Automatic CPU speed & power optimizer for Linux.
Jan 07 18:31:17 orpheus systemd[1]: auto-cpufreq.service: Main process exited, code=exited, status=1/FAILURE
Jan 07 18:31:17 orpheus systemd[1]: auto-cpufreq.service: Failed with result 'exit-code'.
```

AdnanHodzic, 2020-01-08:
Did you install `auto-cpufreq` using AUR package? 

Because yesterday I [added changes that should unify systemd unit deployment](https://github.com/AdnanHodzic/auto-cpufreq/issues/12#issuecomment-571697247) across all distros and I have a feeling AUR package wasn't updated. 

I'll reach out to AUR package maintainer regarding this.

In meantime, can you confirm that everything works fine if you follow instructions how to [install/run it from source](https://github.com/AdnanHodzic/auto-cpufreq#get-auto-cpufreq-source-code)?

crian, 2020-01-08:
AUR package has been updated. Because of the upstream changes please remove and then install the package.

dominicdoty, 2020-01-08:
Yes, package was installed through the AUR with yay.
Wiping and reinstalling didn't fix the issue, 
tried both `pacman -R` and `yay -R` removals, and `pacman` and `yay` installs.

Cloning the git and running `sudo python3 auto-cpufreq.py --install` also doesn't work (the service fails to run after a reboot, but works when manually started). Also note that the service is running successfully immediately after install. The failure only shows up after a reboot.

Also noted during AUR package installation with `yay` or manual `makepkg -si` (does not happen with git clone): Daemon install script throws error
```
cp: cannot stat 'scripts/auto-cpufreq.service': No such file or directory
```

Looks like this occurs at line 18 of auto-cpufreq-install.sh
I'm assuming that the way the AUR package is doing the install is breaking this relative path in the shell script. 
Weirdly, the copy of the service file still gets copied into the systemd/system directory. Not sure where that's happening in the install.

crian, 2020-01-08:
The install script  can not copy the script because it is packed in the package. So it is a different path.
But the systemd service gets copied during the install process as you can see in the PKGBUILD.

`install -Dm644 auto-cpufreq.service "$pkgdir/etc/systemd/system/auto-cpufreq.service"`

The AUR package is working fine on my end even after reboot. So it must a fault on your end which is causing issue.

dominicdoty, 2020-01-08:
Hm. I'm having trouble digging for possible causes because of the vague failure message in the systemd log.

In the mean time, I was able to force it to work by adding the following to the `[Service]` section of the systemd service file.
```
Restart=on-failure
RestartSec=5s
```

The service still fails on first execution after boot but the second execution succeeds.

AdnanHodzic, 2020-03-19:
What happens if you run [auto-cpufreq-installer](https://github.com/AdnanHodzic/auto-cpufreq#installing-auto-cpufreq)?

I also released [ `auto-cpufreq` to Snap store](https://snapcraft.io/auto-cpufreq/) which would definitely fix your problem. But as [I stated for full functionality](https://github.com/AdnanHodzic/auto-cpufreq#snap-store) `auto-cpufreq` needs snapd version 2.4.4 which is currently only available in Ubuntu 20.04.

dominicdoty, 2020-03-19:
I'm not sure where along the line this was resolved, but checking my logs now it appears this is no longer an issue. The service runs correctly after reboots (with my hack fix on the systemd service removed also).
I guess feel free to close at this point?

closed 2020-03-20 by AdnanHodzic