AdnanHodzic/auto-cpufreq #31: auto-cpufreq --log seemn

validatedev, 2020-01-11:


validatedev, 2020-01-11:
Output of `systemctl status auto-cpufreq`
```
● auto-cpufreq.service - auto-cpufreq - Automatic CPU speed & power optimizer f>
     Loaded: loaded (/etc/systemd/system/auto-cpufreq.service; enabled; vendor >
     Active: active (running) since Sat 2020-01-11 11:34:22 +03; 52min ago
   Main PID: 1811 (python3)
      Tasks: 1 (limit: 9195)
     Memory: 17.8M
     CGroup: /system.slice/auto-cpufreq.service
             └─1811 /usr/bin/python3 /usr/bin/auto-cpufreq --daemon

```

validatedev, 2020-01-11:
FYI, `sudo python3 auto-cpufreq.py --log` also seems stopped.

Exorust, 2020-01-11:
This is possibly because of saving the .service file in /lib directory instead of the /usr/lib directory.

validatedev, 2020-01-11:
> This is possibly because of saving the .service file in /lib directory instead of the /usr/lib directory.

Do you have have a source for that, or just an intuition?

Exorust, 2020-01-12:
Unfortunately only an intuition, I can't track down where the log is being saved. The logging info is being saved. I can verify that through, `systemctl status auto-cpufreq -n20`. But still `auto-cpufreq --log` doesn't work. 

Apart from that, I have also experimented with changing the `service` file and making the following edits,
```
StandardOutput=append:/var/log/auto-cpufreq.log
```
Still no result.

AdnanHodzic, 2020-01-12:
@Exorust can you update me [with what I suggested](https://github.com/AdnanHodzic/auto-cpufreq/issues/6#issuecomment-573389224) in #6

Exorust, 2020-01-12:
Its empty

AdnanHodzic, 2020-01-12:
Then something is terribly wrong. Can you tell me what distro are you using, along with Python version and version for psutil, click, power packages.

validatedev, 2020-01-12:
That command sometimes freezes, sometimes works. Interesting.

validatedev, 2020-01-12:
We can do something with the changes on the .py file. Will trying to redirect the stdout to log file directly.

validatedev, 2020-01-12:
@AdnanHodzic Could you reproduce the issue with `systemctl restart auto-cpufreq.service` command?

validatedev, 2020-01-12:
@Exorust Could you have a chance for trying the PR?

Exorust, 2020-01-13:
> Then something is terribly wrong. Can you tell me what distro are you using, along with Python version and version for psutil, click, power packages.

I'm using Ubuntu 16.04, psutil 5.4.8, click 7.0, power 1.4 and python 3.5.2

AdnanHodzic, 2020-01-15:
Can everyone who has this problem, run following:

```
sudo auto-cpufreq --remove
git checkout -b validatedev-fix-file-log master
git pull https://github.com/validatedev/auto-cpufreq.git fix-file-log
sudo python3 auto-cpufreq.py --install
auto-cpufreq --log
```

and see if you still have logs stopping after suspend/hibernate.

smarambas, 2020-01-26:
@AdnanHodzic I had the same problem with the logs freezing but after applying @validatedev's fix it seems to work again after suspend/hibernate. Just to know, I'm using Fedora 31, python 3.7.6, psutil 5.6.3, click 7.0 and power 1.4.
Anyway, thanks for this nice tool.

paulhewson, 2020-03-15:
> Can everyone who has this problem, run following:
>
> ```
> sudo auto-cpufreq --remove
> git checkout -b validatedev-fix-file-log master
> git pull https://github.com/validatedev/auto-cpufreq.git [...]

Hi, I have the same issue with logs here (Ubuntu 19.10 - python3 3.7.5-1).
Tried the fix you're suggesting but I have some problems with the validatedev git repo

```
alessandro@Ale:~$ git checkout -b validatedev-fix-file-log master
fatal: not a git repository (or any of the parent directories): .git
alessandro@Ale:~$ git pull https://github.com/validatedev/auto-cpufreq.git fix-file-log
fatal: not a git repository (or any of the parent directories): .git
```

validatedev, 2020-03-15:
> Hi, I have the same issue with logs here (Ubuntu 19.10 - python3 3.7.5-1).
> Tried the fix you're suggesting but I have some problems with the validatedev git repo
>
> ```
> alessandro@Ale:~$ git checkout [...]

I deleted the repo. Let me create again.

validatedev, 2020-03-15:
@paulhewson and everyone, please try this:
```
sudo auto-cpufreq --remove
mkdir validatedev-auto-cpufreq
cd validatedev-auto-cpufreq
git clone https://github.com/validatedev/auto-cpufreq.git
cd auto-cpufreq
sudo python3 auto-cpufreq.py --install
auto-cpufreq --log
```

paulhewson, 2020-03-15:
Ok, thanks, I have downloaded your repo, so now I'm going to suspend my laptop, resume it and see if logs for auto-cpufreq work.
One curiosity: so your repo is basically this auto-cpufreq with the addition of this fix for log files, isn't it?

validatedev, 2020-03-15:
> Ok, thanks, I have downloaded your repo, so now I'm going to suspend my laptop, resume it and see if logs for auto-cpufreq work.
> One curiosity: so your repo is basically this auto-cpufreq with the addition of this fix for log files, isn't it?

Yes, it is just this. In addition, in new version that fix has been merged (in snap branch). You can safely continue to use my repo until the new version of the program appears.

AdnanHodzic, 2020-03-16:
@validatedev thank you for your assistance. 

@paulhewson I've merged all the changes which were made as part of `snap` package into `master`. Please pull the latest changes.

Since this problem is now fixed, I'll close this bug report.

closed 2020-03-16 by AdnanHodzic