AdnanHodzic/auto-cpufreq #33: ARM Devices

swetoast, 2020-01-12:
```
Traceback (most recent call last):
  File "auto-cpufreq.py", line 448, in <module>
    cli()
  File "/usr/local/lib/python3.7/dist-packages/click/core.py", line 764, in __call__
    return self.main(*args, **kwargs)
  File "/usr/local/lib/python3.7/dist-packages/click/core.py", line 717, in main
    rv = self.invoke(ctx)
  File "/usr/local/lib/python3.7/dist-packages/click/core.py", line 956, in invoke
    return ctx.invoke(self.callback, **ctx.params)
  File "/usr/local/lib/python3.7/dist-packages/click/core.py", line 555, in invoke
    return callback(*args, **kwargs)
  File "auto-cpufreq.py", line 421, in cli
    sysinfo()
  File "auto-cpufreq.py", line 349, in sysinfo
    print("CPU" + str(core_num) + " temp: {:.0f}".format(core_temp['coretemp'][core_num].current) + "°C")
KeyError: 'coretemp'
```

Seems that it can read the CPU temp easily fixed by adding 

cat /sys/devices/virtual/thermal/thermal_zone0/temp 
cat /etc/armbianmonitor/datasources/soctemp

For checking temp if not available via the pypi packages :) 

anyways thanks for a great tool

AdnanHodzic, 2020-01-12:
Thanks, I'll add this on ToDo list for the next release.

Red-Eyed, 2020-08-06:
@swetoast could you please test the new release, as I this bug should be solved

swetoast, 2020-08-09:
works, closing issue tnx for the works guys

closed 2020-08-09 by swetoast

AdnanHodzic, 2020-08-09:
It's too late to announce ARM support as part of 1.2 release now, but it'll definitely be in release for next release.