AdnanHodzic/auto-cpufreq #37: Error starting

andreacocci95, 2020-01-24:
Hi, when I try to run your software I receive this error message:

```
andrea@andrea-X510UQR:~/.tmp/auto-cpufreq$ sudo python3 auto-cpufreq.py
 Traceback (most recent call last):
  File "auto-cpufreq.py", line 32, in <module>
    gov_state = get_cur_gov.split()[0]
IndexError: list index out of range
```

How can I resolve this? Thanks in advance!

AdnanHodzic, 2020-01-25:
Can you verify you're running all the latest Python packages, i.e:
```
sudo pip3 install psutil click distro power
```

andreacocci95, 2020-01-25:
I receive this message after the send your command
```
Requirement already satisfied: psutil in /usr/lib/python3/dist-packages (5.5.1)
Requirement already satisfied: click in /usr/lib/python3/dist-packages (7.0)
Requirement already satisfied: distro in /usr/lib/python3/dist-packages (1.3.0)
Requirement already satisfied: power in /usr/lib/python3/dist-packages (1.4)
```

AdnanHodzic, 2020-03-19:
What happens if you run [auto-cpufreq-installer](https://github.com/AdnanHodzic/auto-cpufreq#installing-auto-cpufreq)?

I also released [ `auto-cpufreq` to Snap store](https://snapcraft.io/auto-cpufreq/) which would definitely fix your problem. But as [I stated for full functionality](https://github.com/AdnanHodzic/auto-cpufreq#snap-store) `auto-cpufreq` needs snapd version 2.4.4 which is currently only available in Ubuntu 20.04.

Red-Eyed, 2020-08-05:
@AdnanHodzic this error means that `get_cur_gov.split()` returns empty list. i.e no governors at all.
is that possible?

AdnanHodzic, 2020-08-05:
@Red-Eyed I can't remember seeing anything like that, but it could be.

I'll need to refactor this part with exception handling (try, except). So if this is really the case at least the whole app doesn't crash.

Red-Eyed, 2020-08-05:
@AdnanHodzic if no governors are available, than we should just call `sys.exit()`?
Because the point of the project is switching across governors:)

AdnanHodzic, 2020-08-05:
@Red-Eyed agreed, no point in proceeding if no governors are available :)

AdnanHodzic, 2020-08-06:
Many new changes have been done as part of 1.2 release and I believe your problem is solved. 

Install latest version using snap package:

```
sudo snap install auto-cpufreq
```

or pull the latest changes and run:

```
sudo ./auto-cpufreq-installer
```

In case this doesn't solve your problem, please feel free to re-open this issue.

closed 2020-08-06 by AdnanHodzic