AdnanHodzic/auto-cpufreq #39: Import error when launching 'sudo auto-cpufreq --install'

Incedis, 2020-02-10:
Greetings, 

OS: Ubuntu 19.10

Installed from source. Added required packages with PIP. 
Tried to install and got error below

Traceback (most recent call last):
  File "/usr/bin/auto-cpufreq", line 11, in <module>
    import psutil
  File "/usr/local/lib/python3.7/dist-packages/psutil/__init__.py", line 96, in <module>
    from . import _pslinux as _psplatform
  File "/usr/local/lib/python3.7/dist-packages/psutil/_pslinux.py", line 26, in <module>
    from . import _psutil_linux as cext
ImportError: /usr/local/lib/python3.7/dist-packages/psutil/_psutil_linux.cpython-37m-x86_64-linux-gnu.so: failed to map segment from shared object

Any help is much appreciated

AdnanHodzic, 2020-03-19:
What happens if you run [auto-cpufreq-installer](https://github.com/AdnanHodzic/auto-cpufreq#installing-auto-cpufreq)?

I also released [ `auto-cpufreq` to Snap store](https://snapcraft.io/auto-cpufreq/) which would definitely fix your problem. But as [I stated for full functionality](https://github.com/AdnanHodzic/auto-cpufreq#snap-store) `auto-cpufreq` needs snapd version 2.4.4 which is currently only available in Ubuntu 20.04.

q-codes, 2020-04-08:
`pip3 install setuptools distro click power psutils
python3 setup.py install`
Finally `./bin/auto-cpufreq`

AdnanHodzic, 2020-08-06:
Many new changes have been implemented as part of 1.2 release and I believe your problem is solved. 

Install latest version using snap package:

```
sudo snap install auto-cpufreq
```

or pull the latest changes and run:

```
sudo ./auto-cpufreq-installer
```

In case this doesn't solve your problem, please feel free to re-open this issue.

closed 2020-08-06 by AdnanHodzic