AdnanHodzic/auto-cpufreq #43: failed to install from aur

Wrizz, 2020-03-19:
install: cannot stat 'auto-cpufreq.py': No such file or directory
==> ERROR: A failure occurred in package().
    Aborting...
Error making: auto-cpufreq-git

I got this error, is there a fix?

AdnanHodzic, 2020-03-19:
@crian is the maintainer of AUR package, so could you please both gives an update? Also, I remember somewhere you mentioned it would be if there were releases, now [there are as I released v1.0](https://github.com/AdnanHodzic/auto-cpufreq/releases/tag/1.0).

@Wrizz until there's an update to AUR package, I'd suggest you to use [auto-cpufreq-installer](https://github.com/AdnanHodzic/auto-cpufreq#installing-auto-cpufreq) and please inform me if you get any issues.

I also released [ `auto-cpufreq` to Snap store.](https://snapcraft.io/auto-cpufreq/) But since I see latest version of [snapd in AUR is 2.43](https://aur.archlinux.org/packages/snapd/), and as [I stated for full functionality](https://github.com/AdnanHodzic/auto-cpufreq#snap-store) `auto-cpufreq` needs snapd version 2.4.4 which is currently only available in Ubuntu 20.04.

crian, 2020-03-23:
AUR package has been updated to the latest upstream changes.

AdnanHodzic, 2020-03-23:
@crian great, thank you!

@Wrizz please retry by installing latest version of [auto-cpufreq AUR](https://aur.archlinux.org/packages/?O=0&SeB=nd&K=auto-cpufreq&outdated=&SB=n&SO=a&PP=50&do_Search=Go).

Wrizz, 2020-03-23:
thanks, it installed without problems :)

AdnanHodzic, 2020-03-23:
Great to hear, in that closing the issue :)

closed 2020-03-23 by AdnanHodzic