AdnanHodzic/auto-cpufreq #46: /sys/devices/system/cpu/cpu0/cpufreq/energy_performance_preference

volnes, 2020-03-24:
Hi!
Got error when use 
`sudo auto-cpufreq --live`

> Setting to use "performance" governor
> /usr/bin/cpufreqctl: row 267: /sys/devices/system/cpu/cpu0/cpufreq/energy_performance_preference: Access denied
> /usr/bin/cpufreqctl: row 267: /sys/devices/system/cpu/cpu1/cpufreq/energy_performance_preference: Access denied
> /usr/bin/cpufreqctl: row 267: /sys/devices/system/cpu/cpu2/cpufreq/energy_performance_preference: Access denied
> /usr/bin/cpufreqctl: row 267: /sys/devices/system/cpu/cpu3/cpufreq/energy_performance_preference: Access denied

I check that files and they doesn't exist....
Ubuntu 18.04.4

AdnanHodzic, 2020-03-24:
Did you use `auto-cpufreq-installer` to install?

volnes, 2020-03-24:
> auto-cpufreq-installer
this

AdnanHodzic, 2020-03-24:
@validatedev any idea on this one? Seems to be related to [latest changes to `cpufreqctl` file.](https://github.com/AdnanHodzic/auto-cpufreq/pull/45)

validatedev, 2020-03-24:
@AdnanHodzic Will look at it asap.

validatedev, 2020-03-24:
@volnes Could you have a chance for letting the CPU model to us please?

volnes, 2020-03-24:
Intel i5-2540M @ 3.300GHz

validatedev, 2020-03-24:
I suspect that older CPU's are not support EPP, which is Intel SpeedShift on Windows. Making some changes and let you try @volnes

validatedev, 2020-03-24:
I wanna sure about the absence of the parameters, @volnes could you try `cat /sys/devices/system/cpu/cpu0/cpufreq/energy_performance_available_preferences` please?

volnes, 2020-03-24:
> cat /sys/devices/system/cpu/cpu0/cpufreq/energy_performance_available_preferences

as i say early, there no such file...

[image]

AdnanHodzic, 2020-03-24:
@validatedev perhaps the simplest thing to do (altough it's bit of a hack), would be to add a file check for:

- `/sys/devices/system/cpu/cpu0/cpufreq/energy_performance_available_preferences`
- `/sys/devices/system/cpu/cpu0/cpufreq/energy_performance_available_preferences`

If one of the  files isn't there simply skip run of `s.run("cpufreqctl --epp --set=balance_power", shell=True) ` in `core.py`.

Unless you have some other more elegant solution :)

validatedev, 2020-03-24:
@AdnanHodzic yeah that's what I'm doing right now :)

AdnanHodzic, 2020-03-24:
:+1: looking forward to PR then :)

validatedev, 2020-03-24:
@volnes 
Could you try https://github.com/validatedev/auto-cpufreq/tree/fix-conflict-with-TLP because I want to sure about my fix is working.

volnes, 2020-03-24:
@validatedev 
[image]
i think all ok, no errors

AdnanHodzic, 2020-03-24:
@validatedev I made some comments as there are code conflicts in #47 

Could you please look into them as I'd like to merge these changes ASAP as it's a serious bug and I want to release new version of snap including these changes as well.

Thanks

AdnanHodzic, 2020-03-24:
@volnes problem has been fixed as part of #48 PR. Please pull the latest changes and re-install the tool. 

Please feel free to re-open the issue if you're still experiencing the same issue.

closed 2020-03-24 by AdnanHodzic

volnes, 2020-03-25:
@AdnanHodzic @validatedev  yes update, problem fixed, thanks guys for your work