AdnanHodzic/auto-cpufreq #49: AUR package not working with python virtualenv

marc0der, 2020-03-25:
To bring stability to my system, I use virtualenv to install all python tools through package managers like pip. When installing the `auto-cpufreq-git` package from the AUR, I noticed that it installed into my virtualenv in user space:

    $ which auto-cpufreq
    /home/marco/.virtualenv/bin/auto-cpufreq

When `auto-cpufreq` is started, it borks with the following:

```
● auto-cpufreq.service - auto-cpufreq - Automatic CPU speed & power optimizer for Linux
     Loaded: bad-setting (Reason: Unit auto-cpufreq.service has a bad unit file setting.)
     Active: inactive (dead)

Mar 24 11:09:21 rocinante systemd[1]: /usr/lib/systemd/system/auto-cpufreq.service:9: Executable "auto-cpufreq" not found in path "/usr/local/bin:/usr/bin"
Mar 24 11:09:21 rocinante systemd[1]: auto-cpufreq.service: Unit configuration has fatal error, unit will not be started.
Mar 24 11:10:12 rocinante systemd[1]: /usr/lib/systemd/system/auto-cpufreq.service:9: Executable "auto-cpufreq" not found in path "/usr/local/bin:/usr/bin"
Mar 24 11:10:12 rocinante systemd[1]: auto-cpufreq.service: Unit configuration has fatal error, unit will not be started.
```

AdnanHodzic, 2020-03-25:
It seems to be this issue is very similar to #42 

Could you please [try doing what I suggest here](https://github.com/AdnanHodzic/auto-cpufreq/issues/42#issuecomment-603714391), in case it also works for you I'll make necessary changes to permanently resolve this problem.

crian, 2020-03-27:
He is manipulating the install path, so the daemon can't find the file. It expects the file in `usr/local/bin` or `usr/bin`.

marc0der, 2020-03-27:
@crian well, to be fair it isn't me manipulating the install path. It is virtualenv. And this will happen to anyone using virtualenv or pyenv would be my guess.

crian, 2020-03-27:
@marc0der can you `git clone https://aur.archlinux.org/auto-cpufreq-git.git` and edit `python setup.py install --root="${pkgdir}"` to `python setup.py install --prefix=/usr --root="${pkgdir}"` in the `PKGBUILD` and install it?

marc0der, 2020-03-27:
@crian I tried your suggestion but it didn't work either. I've been messing around for over an hour trying various things to get it to work, but to no avail. It seems that many paths are hardcoded in the package and service file itself, that simply don't align with Arch. Then there is a problem of unresolved dependencies too. Calling it a night now...

crian, 2020-03-28:
@marc0der need more details

AdnanHodzic, 2020-03-29:
@marc0der snapd 2.44 is now [available in AUR](https://aur.archlinux.org/packages/snapd/) please install it and then give [auto-cpufreq snap package](https://snapcraft.io/auto-cpufreq) a try `sudo snap install auto-cpufreq`

marc0der, 2020-03-29:
@AdnanHodzic thanks for that, I've just installed it with snap and am now running into some other errors. I'll open a new issue for that.

@crian sorry for not pursuing the AUR package any further, feel free to close this issue if you see fit.

AdnanHodzic, 2020-03-29:
@marc0der sorry to hear that. There will be errors if your snap/d version is < 2.44. But either way, looking forward to hearing more.

AdnanHodzic, 2020-08-06:
Many new changes have been implemented as part of 1.2 release and there's an improved snap package.

```
sudo snap install auto-cpufreq
```

I also asked @crian if he could build a new [AUR package based on 1.2 release](https://github.com/AdnanHodzic/auto-cpufreq/releases/tag/1.2).

closed 2020-08-06 by AdnanHodzic

dausruddin, 2021-06-16:
Just want to notify that this issue is still happening. I ended up uninstalling pyenv to install auto-cpufreq, then reinstall pyenv back. I installed via AUR.