AdnanHodzic/auto-cpufreq #51: Daemon cannot start on Manjaro

marc0der, 2020-03-29:
I've installed auto-cpufreq on my up-to-date Manjaro machine with snap:

```
snap     2.44.1-1
snapd    2.44-1
series   16
manjaro  -
kernel   4.19.112-1-MANJARO
```

Installation goes well, but things go wrong when I try to start the daemon. This is from system logs:

```
Mar 29 14:56:22 rocinante auto-cpufreq.service[1881264]: ----------------------------- System information ------------------------------
Mar 29 14:56:22 rocinante auto-cpufreq.service[1881264]: Traceback (most recent call last):
Mar 29 14:56:22 rocinante auto-cpufreq.service[1881264]:   File "/snap/auto-cpufreq/5/bin/auto-cpufreq", line 108, in <module>
Mar 29 14:56:22 rocinante auto-cpufreq.service[1881264]:     main()
Mar 29 14:56:22 rocinante auto-cpufreq.service[1881264]:   File "/snap/auto-cpufreq/5/lib/python3.6/site-packages/click/core.py", line 829, in __call__
Mar 29 14:56:22 rocinante auto-cpufreq.service[1881264]:     return self.main(*args, **kwargs)
Mar 29 14:56:22 rocinante auto-cpufreq.service[1881264]:   File "/snap/auto-cpufreq/5/lib/python3.6/site-packages/click/core.py", line 782, in main
Mar 29 14:56:22 rocinante auto-cpufreq.service[1881264]:     rv = self.invoke(ctx)
Mar 29 14:56:22 rocinante auto-cpufreq.service[1881264]:   File "/snap/auto-cpufreq/5/lib/python3.6/site-packages/click/core.py", line 1066, in invoke
Mar 29 14:56:22 rocinante auto-cpufreq.service[1881264]:     return ctx.invoke(self.callback, **ctx.params)
Mar 29 14:56:22 rocinante auto-cpufreq.service[1881264]:   File "/snap/auto-cpufreq/5/lib/python3.6/site-packages/click/core.py", line 610, in invoke
Mar 29 14:56:22 rocinante auto-cpufreq.service[1881264]:     return callback(*args, **kwargs)
Mar 29 14:56:22 rocinante auto-cpufreq.service[1881264]:   File "/snap/auto-cpufreq/5/bin/auto-cpufreq", line 39, in main
Mar 29 14:56:22 rocinante auto-cpufreq.service[1881264]:     sysinfo()
Mar 29 14:56:22 rocinante auto-cpufreq.service[1881264]:   File "/snap/auto-cpufreq/5/lib/python3.6/site-packages/source/core.py", line 360, in sysinfo
Mar 29 14:56:22 rocinante auto-cpufreq.service[1881264]:     dist = distro + " " + version
Mar 29 14:56:22 rocinante auto-cpufreq.service[1881264]: UnboundLocalError: local variable 'version' referenced before assignment
```

I did a bit of digging and found that the _core.py_ script is failing due to not being able to locate the `searchfile` at `/var/lib/snapd/hostfs/etc/os-release`. The bug here seems that it assumes the file is always present (which it seems not on my system), and if the file is not present then the `version` and `distro` variables go uninitialised. Looking at that path on my machine, I have an empty directory at `/var/lib/snapd/hostfs`.

Red-Eyed, 2020-08-05:
I fixed this in #62 PR, not sure if that fix went to snap

Red-Eyed, 2020-08-05:
Now it should work fine

AdnanHodzic, 2020-08-05:
I want to first fix `snap` package and make a new release. I want to make sure it's working for @marc0der as he waited long enough for this :)

Red-Eyed, 2020-08-05:
Yeah, IMHO: snap is mandatory for now, as most of people use ubuntu and they don't want to install it manually (from github)
I could help to test it for POP OS!

AdnanHodzic, 2020-08-05:
Will, let you know once the problem is fixed and release has been made.

AdnanHodzic, 2020-08-06:
@marc0der could you please try new (unreleased version) in beta, and confirm if you still have the same problem? 

```
sudo snap install auto-cpufreq --beta
```

marc0der, 2020-08-06:
Just tested and seems to be working fine now! :tada: Thanks @Red-Eyed!

closed 2020-08-06 by marc0der