AdnanHodzic/auto-cpufreq #53: Snap deamon is polluting syslog

Kwaadpepper, 2020-04-13:
[image]

Is there anyway to silent the deamon ?

AdnanHodzic, 2020-08-06:
Many new changes have been implemented as part of 1.2 release and I believe your problem is solved. 

Install latest version using snap package:

```
sudo snap install auto-cpufreq
```

or pull the latest changes and run:

```
sudo ./auto-cpufreq-installer
```

In case this doesn't solve your problem, please feel free to re-open this issue.

Kwaadpepper, 2020-08-07:
Sorry to say that this is not solved.
#81 
I have posted my syslog

AdnanHodzic, 2020-08-07:
Seems like this issue is identical what was reported in #82

Red-Eyed, 2020-09-02:
Any news about that?
I had to disable auto-cpufreq as it prints too much to my syslog
I thin that we should increase the priority of this task

AdnanHodzic, 2020-09-02:
I'm swamped IRL, but I'll do my best to take a look at this tonight.

Seems like this issue is only present on snap environemnts, as I couldn't replicate the problem when `auto-cpufreq` was installed using `auto-cpufreq-install` installer.

AdnanHodzic, 2020-09-02:
Released a fix as part of 1.2.3 version which is available in beta channel, install it by running:

```
sudo snap install auto-cpufreq --beta
```

After you confirm it's working as expected for you I'll promote it to stable channel, merge changes with master and close this issue.

Kwaadpepper, 2020-09-02:
Thanks I'll check this ASAP.

closed 2020-09-02 by AdnanHodzic

AdnanHodzic, 2020-09-02:
Fix has already been confirmed in #82. Please feel to re-open if the problem persists. Changes have also been promoted to stable channel, so in case you switched to beta, you can switch back to stable by running:

```
sudo snap switch --channel=stable auto-cpufreq
```

commit 566fd4a7 (2020-09-02): Revamp of writing daemon service logs for snap envs (Close: #53, #82)