AdnanHodzic/auto-cpufreq #55: Permission denied with snap

wendersonj, 2020-04-27:
Hello ! I'm testing this program, but so far, so good. Although, I found a problem and resolved it.
When I first installed it with 'snap', when running with 'sudo the auto-cpufreq --live', I was getting permission denied when trying to make changes.
I removed it and re-installed using the source code. Then, everything changed as supposed too. 

[my small contribution, even though I don't now if it's a know issue...]
Cheers!

wendersonj, 2020-04-27:
Using Ubuntu 18.04 bionic with kernel 4.15.0-96-generic; Intel i7-8565U.

closed 2020-04-27 by wendersonj

AdnanHodzic, 2020-04-27:
Thank you. As I [indicated due to changes that needed to be done to snapd](https://github.com/AdnanHodzic/auto-cpufreq#snap-store). `snap` package is currently only encouraged on 20.04.