AdnanHodzic/auto-cpufreq #59: Permission denied on Snap 2.44.3+20.04

Booteille, 2020-05-21:
Hi!

Using `auto-cpufreq --log" after a successful "sudo auto-cpufreq --install", I get the following error:

```console
⋊> ~ auto-cpufreq --log

-------------------------------- Daemon check ---------------------------------

ERROR:

Daemon not enabled, must run install first, i.e:
sudo auto-cpufreq --install

-------------------------------------------------------------------------------

----------------------------- System information ------------------------------

Traceback (most recent call last):
  File "/snap/auto-cpufreq/5/bin/auto-cpufreq", line 108, in <module>
    main()
  File "/snap/auto-cpufreq/5/lib/python3.6/site-packages/click/core.py", line 829, in __call__
    return self.main(*args, **kwargs)
  File "/snap/auto-cpufreq/5/lib/python3.6/site-packages/click/core.py", line 782, in main
    rv = self.invoke(ctx)
  File "/snap/auto-cpufreq/5/lib/python3.6/site-packages/click/core.py", line 1066, in invoke
    return ctx.invoke(self.callback, **ctx.params)
  File "/snap/auto-cpufreq/5/lib/python3.6/site-packages/click/core.py", line 610, in invoke
    return callback(*args, **kwargs)
  File "/snap/auto-cpufreq/5/bin/auto-cpufreq", line 39, in main
    sysinfo()
  File "/snap/auto-cpufreq/5/lib/python3.6/site-packages/source/core.py", line 351, in sysinfo
    searchfile = open("/var/lib/snapd/hostfs/etc/os-release", "r")
PermissionError: [Errno 13] Permission denied: '/var/lib/snapd/hostfs/etc/os-release'
```

Tried the command with `sudo`, same result.

AdnanHodzic, 2020-05-23:
Could you please tell me which distribution/release are you running?

Booteille, 2020-05-23:
Sure!

Pop_OS! 20.04

AdnanHodzic, 2020-05-23:
Will look into this. Until I resolve this issue, I would suggest to use [auto-cpufreq-installer](https://github.com/AdnanHodzic/auto-cpufreq#auto-cpufreq-installer)

Booteille, 2020-05-23:
That's what I finally did.

Thank you!

AdnanHodzic, 2020-08-06:
I would be interested in hearing if you still have this problem with snap as as part of new (unreleased) release/beta:

```
sudo snap install auto-cpufreq --beta
```

Booteille, 2020-08-06:
Hi! It seems to work properly now. Thanks!

closed 2020-08-06 by AdnanHodzic

Red-Eyed, 2020-08-06:
@AdnanHodzic Unfortunately, it still throws the same error on POP OS 20.04

AdnanHodzic, 2020-08-06:
@Red-Eyed which version of `snapd` do you have?

It's weird that it's still happening for you and not for @Booteille

Could you try running following and reporting if same problem is still there:
```
sudo snap remove --purge auto-cpufreq
sudo snap install auto-cpufreq --beta
```

Red-Eyed, 2020-08-06:
Yeah, it's weird
```
➜  ~ sudo snap remove --purge auto-cpufreq
[sudo] password for redeyed: 
auto-cpufreq removed
➜  ~ sudo snap remove --purge auto-cpufreq
➜  ~ sudo snap install auto-cpufreq --beta
auto-cpufreq (beta) 1.2 from Adnan Hodzic (fooctrl) installed
➜  ~ sudo auto-cpufreq                   

-------------------------------- auto-cpufreq ---------------------------------

Automatic CPU speed & power optimizer for Linux

Example usage:
auto-cpufreq --monitor

-----

Usage: auto-cpufreq [OPTIONS]

Options:
  --monitor             Monitor and suggest CPU optimizations
  --live                Monitor and make suggested CPU optimizations
  --install / --remove  Install/remove daemon for automatic CPU optimizations
  --log                 View live CPU optimization log made by daemon
  --help                Show this message and exit.

-------------------------------------------------------------------------------

➜  ~ sudo auto-cpufreq --monitor

----------------------------- System information ------------------------------

Traceback (most recent call last):
  File "/snap/auto-cpufreq/6/bin/auto-cpufreq", line 111, in <module>
    main()
  File "/snap/auto-cpufreq/6/lib/python3.6/site-packages/click/core.py", line 829, in __call__
    return self.main(*args, **kwargs)
  File "/snap/auto-cpufreq/6/lib/python3.6/site-packages/click/core.py", line 782, in main
    rv = self.invoke(ctx)
  File "/snap/auto-cpufreq/6/lib/python3.6/site-packages/click/core.py", line 1066, in invoke
    return ctx.invoke(self.callback, **ctx.params)
  File "/snap/auto-cpufreq/6/lib/python3.6/site-packages/click/core.py", line 610, in invoke
    return callback(*args, **kwargs)
  File "/snap/auto-cpufreq/6/bin/auto-cpufreq", line 66, in main
    sysinfo()
  File "/snap/auto-cpufreq/6/lib/python3.6/site-packages/source/core.py", line 400, in sysinfo
    searchfile = open("/var/lib/snapd/hostfs/etc/os-release", "r")
PermissionError: [Errno 13] Permission denied: '/var/lib/snapd/hostfs/etc/os-release'
➜  ~ sudo auto-cpufreq --debug  
Usage: auto-cpufreq [OPTIONS]
Try 'auto-cpufreq --help' for help.

Error: no such option: --debug
➜  ~ snap --version
snap    2.45.2
snapd   2.45.2
series  16
pop     20.04
kernel  5.4.0-7634-generic
➜  ~ 
➜  ~ sudo auto-cpufreq --debug
➜  ~ sudo snap run auto-cpufreq --debug
Usage: auto-cpufreq [OPTIONS]
Try 'auto-cpufreq --help' for help.

Error: no such option: --debug
➜  ~ sudo snap run auto-cpufreq --monitor

----------------------------- System information ------------------------------

Traceback (most recent call last):
  File "/snap/auto-cpufreq/6/bin/auto-cpufreq", line 111, in <module>
    main()
  File "/snap/auto-cpufreq/6/lib/python3.6/site-packages/click/core.py", line 829, in __call__
    return self.main(*args, **kwargs)
  File "/snap/auto-cpufreq/6/lib/python3.6/site-packages/click/core.py", line 782, in main
    rv = self.invoke(ctx)
  File "/snap/auto-cpufreq/6/lib/python3.6/site-packages/click/core.py", line 1066, in invoke
    return ctx.invoke(self.callback, **ctx.params)
  File "/snap/auto-cpufreq/6/lib/python3.6/site-packages/click/core.py", line 610, in invoke
    return callback(*args, **kwargs)
  File "/snap/auto-cpufreq/6/bin/auto-cpufreq", line 66, in main
    sysinfo()
  File "/snap/auto-cpufreq/6/lib/python3.6/site-packages/source/core.py", line 400, in sysinfo
    searchfile = open("/var/lib/snapd/hostfs/etc/os-release", "r")
PermissionError: [Errno 13] Permission denied: '/var/lib/snapd/hostfs/etc/os-release'
➜  ~  
```

Red-Eyed, 2020-08-06:
Also, I don't see `--debug` option
but AFAIK it should be present, isn't it?

AdnanHodzic, 2020-08-06:
> Also, I don't see `--debug` option
> but AFAIK it should be present, isn't it?

You're right, I just pushed another revision.

Could you retry running: 

```
sudo snap remove --purge auto-cpufreq
sudo snap install auto-cpufreq --beta
```

again?

Red-Eyed, 2020-08-06:
Yeah, I've tried
same error with permission denied. but `--debug` option
also, `inxi` is not installed

Maybe, let's just wrap distribution into try...except block and catch Permission denied
and could you please install inxi in snap?

If you want, I can add handling of PermissionDenied

Red-Eyed, 2020-08-06:
check it out: #75

Red-Eyed, 2020-08-06:
@AdnanHodzic can I build snap locally and install it locally?
I want to find out the problem on pop os

Red-Eyed, 2020-08-06:
Sorry, I'm asking about "how can I test snap builds" ?
maybe you could give me some tutorials?