AdnanHodzic/auto-cpufreq #64: [Question] setting turbo in a powersave mode doesn't seem right

Red-Eyed, 2020-08-05:
Hello @AdnanHodzic ,  
Do we need turbo boost in a powersave mode?

AdnanHodzic, 2020-08-05:
I remember this was raised by @validatedev in one of his PR's. IIRC, reason to leave turbo boost enabled while in`powersave` was due to performance issues. Although it did greatly improve the battery life.

I think this is a great option to have in #26 where users could decide on their own whether they want turbo boost on in `powersave` or not.

Red-Eyed, 2020-08-05:
Got it

closed 2020-08-05 by Red-Eyed

validatedev, 2020-09-03:
EPP could take care of it, that's why I set turboboost on when on load. Of course, the implementation of the settings file could be sweet, but right now it works well I think. @Red-Eyed