AdnanHodzic/auto-cpufreq #71: [Feature] add Flatpak support

Red-Eyed, 2020-08-05:
As far as we know that not all linux distribution uses snap by default.
I think that adding flatpak package would be a nice addition for the RHEL/Fedora/Silverblue/CentOS users and others (clear linux, POP OS, and other who relice on flatpak as default)

I have a bit of experience with flatpak, so can try to look in that direction.

@AdnanHodzic  if you consider flatpak support, than you can assign this issue to me.

AdnanHodzic, 2020-08-06:
Sure, feel to to start working on it. The more ways users have to install the app the better.

jdferron, 2020-08-07:
Agreed, Linux Mint as of LM20 does not support snap packages.   Flatpak support would be awesome!

Red-Eyed, 2020-08-07:
@jdferron Linux mint [does support it](https://snapcraft.io/docs/installing-snapd), it's just Ubuntu/Debian based.
Just don't confuse, you probably wanted to say that `snapd` is not preinstalled by default :)

dlalinsp, 2020-09-02:
@Red-Eyed It is not just not pre-installed by default, though. `snapd` is actively blocked from being installed without running a command first.

 I would also like to say that I think Linux Mint is a great distro and I believe it is a valuable contributor to the Linux community. Therefore, I think it isn't fair to say that they are *just Ubuntu/Debian with whistles.*

mmstick, 2020-09-02:
This kind of software cannot work as a Flatpak. Flatpak is designed for desktop applications, rather than system services, and applications run by it are executed in a separate namespace with limited access to the host. You can't do anything that requires root, without having a DBus system service installed and running on the host that can do what's needed.

AdnanHodzic, 2020-09-02:
Thanks for your confirmation @mmstick as this was my suspicion from the very beginning from my (limited) knowledge on how Flatpak works.

Since this is a "wont fix", I'm closing the issue. If someone has anything else to add, or would like to have further discussions please feel free to re-open it.

closed 2020-09-02 by AdnanHodzic