AdnanHodzic/auto-cpufreq #74: [Feature] add sensors info into --debug

Red-Eyed, 2020-08-06:
Since sensors are very different across the machines I think that it would be nice to add output of `psutil.sensors_temperature()` into `get_sys_info()` function

Red-Eyed, 2020-08-06:
@AdnanHodzic do you have any idea how to get snpad version from inside the snap package?

AdnanHodzic, 2020-08-06:
FWIK, you can't do that. Because basically snap is running inside of a container, hence all the reasons snap environment has all these exceptions within auto-cpufreq, [perfect example](https://github.com/AdnanHodzic/auto-cpufreq/blob/master/source/core.py#L440). 

But since I'm pretty sure there could be a file that could have this information stored, maybe ask this same question on [snapcraft forum](forum.snapcraft.io/)?

AdnanHodzic, 2021-02-06:
Fixed as part of [1.6.0 release](https://github.com/AdnanHodzic/auto-cpufreq/releases/tag/v1.6.0).

closed 2021-02-06 by AdnanHodzic