AdnanHodzic/auto-cpufreq #82: The systemd log is being spammed by auto-cpufreq

marc0der, 2020-08-07:
I've started using the tool and loving it so far. Good work!

After installation, the `auto-cpufreq` service is continually outputting the full payload of the `--log` command to the system logs. 

When tailing the logs with `journalctl --follow`, I see a continual stream of these log reports being dumped in the log every 5 seconds. Should this perhaps rather be a one-line summary whenever the state of cpu scaling is changed by the tool?

I realise that these are all teething issues that still need to be sorted.
Keep up the great work!

AdnanHodzic, 2020-08-07:
First of all thanks :)

I remember this same thing was mentioned in #53. Hence, I think it would be a good think to address this issue in one of the future releases.

marc0der, 2020-08-07:
Great stuff. Some further digging has revealed that all logging is done straight to standard out. Perhaps this issue could be circumvented using targeted logging using a logger library. I could look into this, but since python isn't my first language I'm not promising anything.

AdnanHodzic, 2020-08-07:
Go for it, it would be amazing to see a PR from you :) Otherwise, even if you find a nice library that could be used for this purpose that would also be great.

AdnanHodzic, 2020-09-02:
Released a fix as part of 1.2.3 version which is available in beta channel, install it by running:

```
sudo snap install auto-cpufreq --beta
```

After you confirm it's working as expected for you I'll promote it to stable channel, merge changes with master and close this issue.

marc0der, 2020-09-02:
I just tried it out and it works perfectly :+1:

AdnanHodzic, 2020-09-02:
Awesome, enjoy

closed 2020-09-02 by AdnanHodzic