AdnanHodzic/auto-cpufreq #87: [Feature] Tray-Icon for Status-Info

schinfo, 2020-08-08:
Can you add a tray icon (red / green) so that the status (active / inactive) of auto-cpugreq is visable.

Red-Eyed, 2020-08-08:
@AdnanHodzic I could help with this, maybe even implement. Had experience with python app indicator

Red-Eyed, 2020-08-08:
@AdnanHodzic could please you label issues by category and priority?

schinfo, 2020-08-08:
Only you as the owner can set labels in GitHub ;-)

AdnanHodzic, 2020-08-09:
> @AdnanHodzic I could help with this, maybe even implement. Had experience with python app indicator

Sure, sounds like a nice idea. 

> Only you as the owner can set labels in GitHub ;-)

I just sent @Red-Eyed an invite to be a collaborator, with that status he'll be able to do these things :)

Kwaadpepper, 2020-08-14:
If you do not want to reinvent the wheel there is this tool
https://github.com/konkor/cpufreq

It does not do the automatic job of https://github.com/AdnanHodzic/auto-cpufreq, so both are working well together

schinfo, 2020-08-14:
cpufreq maybe a very good tool, but the tray-icon doesn't work at Cinnamon

yochananmarqos, 2020-11-30:
@Kwaadpepper `cpufreq` conflicts with `auto-cpufreq` as they both install the same `cpufreqctl` binary.

Compare `cpufreq`'s [`cpufreqctl`](https://github.com/konkor/cpufreq/blob/master/cpufreqctl) and `auto-cpufreq`'s [`cpufreqctl.sh`](https://github.com/AdnanHodzic/auto-cpufreq/blob/master/scripts/cpufreqctl.sh).

AdnanHodzic, 2020-12-06:
auto-cpufreq [will take a backup of original ](https://github.com/AdnanHodzic/auto-cpufreq/blob/d0aeb8c2a2efc692e07fee098b9f8757b452591e/source/core.py#L157)`cpufreqctl` if it detects it and will use its own version which has slight improvements over the original.

yochananmarqos, 2020-12-06:
@AdnanHodzic Ah, I see. That will have to be manually dealt with in the AUR package, see my [install file](https://github.com/yochananmarqos/pkgbuilds/blob/master/auto-cpufreq-git/auto-cpufreq.install#L26). I've let the AUR maintainers know.

AdnanHodzic, 2020-12-06:
> @AdnanHodzic Ah, I see. That will have to be manually dealt with in the AUR package, see my [install file](https://github.com/yochananmarqos/pkgbuilds/blob/master/auto-cpufreq-git/auto-cpufreq.install#L26). I've let the AUR maintainers know.

Cool thanks. I think it would be the best if AUR install file did the same thing `auto-cpufreq` is trying to do if `cpufreqctl` is detected, but let's see if @crian has something against this approach.

crian, 2020-12-06:
> Cool thanks. I think it would be the best if AUR install file did the same thing `auto-cpufreq` is trying to do if `cpufreqctl` is detected, but let's see if @crian has something against this.

I'm fine with this. Currently updating the AUR pkg.

AdnanHodzic, 2020-12-06:
> I'm fine with this. Currently updating the AUR pkg.

Nice, thanks. Let me know once this has been done, so I can close this and #133.

crian, 2020-12-06:
> > > > @AdnanHodzic Ah, I see. That will have to be manually dealt with in the AUR package, see my [install file](https://github.com/yochananmarqos/pkgbuilds/blob/master/auto-cpufreq-git/auto-cpufreq.install#L26). I've let the AUR maintainers know.
> > > 
> > > 
> > > Cool thanks. I think it would be the best if AUR install file did the same thing `auto-cpufreq` is trying to do if `cpufreqctl` is detected, but let's see if @crian has something against this.
> > 
> > 
> > I'm fine with this. Currently updating the AUR pkg.
> 
> Nice, thanks. Let me know once this has been done, so I can close part of this discussion and I believe same thing is with #133.

Done https://aur.archlinux.org/cgit/aur.git/commit/?h=auto-cpufreq-git&id=ff0101f0f6552eb085ede88f132d9cb1677f72f1

AdnanHodzic, 2020-12-06:
@crian awesome, thanks! 

@yochananmarqos and everybody else who had problems with `cpufreqctl` AUR conflict, if the problem persist please open a new issue as it got out of scope from the original request. Thanks

yochananmarqos, 2020-12-07:
@AdnanHodzic: Really? Thanks.

> Thanks to our contributors for this release!
>
>@crian

/me clears throat loudly

AdnanHodzic, 2020-12-07:
> @AdnanHodzic: Really? Thanks.
> 
> > Thanks to our contributors for this release!
> > @crian

I like to give people credit for their work. Even if it's as simple as creating new version of AUR package which fixes a bug or 2.

crian, 2020-12-07:
> > @AdnanHodzic: Really? Thanks.
> > > Thanks to our contributors for this release!
> > > @crian
> 
> I like to give people credit for their work. Even if it's as simple as creating new version of AUR package which fixes a bug or 2.

You should give the credit to @yochananmarqos as he fixed it.

AdnanHodzic, 2020-12-07:
@crian thanks for the heads up, I completely missed it. @yochananmarqos sorry about that! Updated [1.5.1 release contributors list](https://github.com/AdnanHodzic/auto-cpufreq/releases/tag/v1.5.1).

AdnanHodzic, 2021-10-16:
After  [v1.7.0 release](https://github.com/AdnanHodzic/auto-cpufreq/releases/tag/v1.7.0) release and [config/settings file going live](https://github.com/AdnanHodzic/auto-cpufreq/blob/master/README.md#configuring-auto-cpufreq). This feature is probably next highest item on my priority list. If anyone wants to give it a try at implementing this feature, I'll make sure to give you credit as part of [releases page](https://github.com/AdnanHodzic/auto-cpufreq/releases).

AdnanHodzic, 2022-11-11:
Since I'm very limited with available time to do this myself, if you or anyone else wants to give it a try in implementing this functionality, please give it a try and [contribute to the project](https://github.com/AdnanHodzic/auto-cpufreq/#code-contribution) and you will be credited for your work as part of [future release](https://github.com/AdnanHodzic/auto-cpufreq/releases).

KAGEYAM4, 2023-09-09:
Any update on this , currently if i want to force switch to peformance mode , opening terminal > sudoing > writing password. A tray indicator will be nice to force switch on the fly ( 3 modes - auto, powersave, perforcmace ). 

Can i use cpu-freq in meantime ? I am on Arch , installed auto-cpu freq thorugh install script. is there any better tray switching method available ?

AdnanHodzic, 2023-09-09:
If you install auto-cpufreq from changes that are on `master` branch right now, you'll get a desktop entry "auto-cpufreq" which allow you to easily switch between performance/powersave/default mode. 

These changes will be released soon with upcoming v2.0 release and I'll make a video & update documentation about new functionality.

KAGEYAM4, 2023-09-09:
> If you install auto-cpufreq from changes that are on `master` branch right now, you'll get a desktop entry "auto-cpufreq" which allow you to easily switch between performance/powersave/default mode using auto-cpufreq GTK GUI app.
> 
> These changes will be released soon with upcoming v2.0 release and I'll make a video & update documentation about new functionality.

when switching mode using gtk gui , it prompts for password , is there a way to skip it ?

AdnanHodzic, 2023-09-09:
> when switching mode using gtk gui , it prompts for password , is there a way to skip it ?

This was done by design. As that's when actual "sudo" privledges will be required in background.

@shadeyg56 now that I think of it, another idea could to require sudo password when auto-cpufreq GUI is initially started. Altough that's not intuative, since maybe some folks will just run it to see the stats.

shadeyg56, 2023-09-09:
right, at some point the GUI HAS to prompt for root password.

Originally, the app actually did prompt during launch, but we changed it because of exactly what you said. I believe it is fine how it is now. 

@KAGEYAM4 because overriding is a root action, it doesn't matter what medium you use, it will always need root somehow. It is quite quick to open up the GUI, select the override and enter your password. Please note that the GUI that is available in master has some known issues that are currently being worked on in another branch, but we hope to merge soon :)

AdnanHodzic, 2023-09-10:
> I believe it is fine how it is now.

It's definitely fine as it is now.

KAGEYAM4, 2023-09-10:
> right, at some point the GUI HAS to prompt for root password.
>
> Originally, the app actually did prompt during launch, but we changed it because of exactly what you said. I believe it is fine how it [...]

Before this i used Windows and KDE, both had quick way to switch power governor, that's what i was hoping here. ( quick switch using tray option )

But seeing that this project whole purpose is to automate this process, i can see that this change won't be used by many.

KAGEYAM4, 2023-09-11:
Is there some other project which provides tray icon to switch power governor and dosen't confilit with `auto-cpufreq` ? 

Or is there some permissions that i can change to add my user/group to not get prompted for sudo access ?

KAGEYAM4, 2023-09-11:
What do you think about this scripts  , will they work correctly ? Original source - [SDDM Guest Login](https://bbs.archlinux.org/viewtopic.php?id=192891)

-------------------------------------------------------------------->

calling a script "guest-account" (see below) in /usr/share/config/kdm/Xsetup. Xsetup is looks like:
```
#! /bin/sh
# Xsetup - run as root before the login dialog appears
/root/scripts/guest-account add
```

And  /usr/share/config/kdm/Xstop looks like:

```
#! /bin/sh
# Xreset - run as root after session exits
if [ $USER = 'guest' ];
then
    /root/scripts/guest-account remove guest
    /root/scripts/guest-account add
fi
```

The script for guest-account:
```
#!/bin/sh -e
# (C) 2008 Canonical Ltd.
# Author: Martin Pitt <martin.pitt@ubuntu.com>
# License: GPL v2 or later
# modified by David D Lowe and Thomas Detoux
#
# Setup user and temporary home directory for guest session.
# If this succeeds, this script needs to print the username as the last line to
# stdout.

add_account ()
{
  mkdir /tmp/guest
  HOME="/tmp/guest"
  USER=`echo $HOME | sed 's/\(.*\)guest/guest/'`

  # if $USER already exists, it must be a locked system account with no existing
  # home directory
  if PWSTAT=`passwd -S "$USER"` 2>/dev/null; then
    if [ "`echo \"$PWSTAT\" | cut -f2 -d\ `" != "L" ]; then
      echo "User account $USER already exists and is not locked"
      exit 1
    fi
    PWENT=`getent passwd "$USER"` || {
      echo "getent passwd $USER failed"
      exit 1
    }
    GUEST_UID=`echo "$PWENT" | cut -f3 -d:`
    if [ "$GUEST_UID" -ge 500 ]; then
      echo "Account $USER is not a system user"
      exit 1
    fi
    HOME=`echo "$PWENT" | cut -f6 -d:`
    if [ "$HOME" != / ] && [ "${HOME#/tmp}" = "$HOME" ] && [ -d "$HOME" ]; then
      echo "Home directory of $USER already exists"
      exit 1
    fi
  else
    # does not exist, so create it

    # Arch Linux modification: Ubuntu/Debian uses their own adduser package,
    # which works differently from the own provided by the shadow package.
    # Instead, use useradd, which works in any distro.
    #
    # Only the syntax is changed
    # adduser          -> useradd
    # --system         -> --system
    # --no-create-home -> --no-create-home
    # --home           -> --home-dir
    # --gecos          -> --comment
    # --group          -> --user-group
    # --shell          -> --shell
    #
    #adduser --system --no-create-home --home / --gecos "Guest" --group --shell /bin/bash $USER || {
    useradd --system --no-create-home --home-dir / --comment "Guest" --user-group --shell /bin/bash $USER || {
        umount "$HOME"
        rm -rf "$HOME"
        exit 1
    }
    echo "guest:guest"|chpasswd
  fi

  # create temporary home directory
  mount -t tmpfs -o mode=700 none "$HOME" || { rm -rf "$HOME"; exit 1; }
  chown $USER:$USER "$HOME"
  gs_skel=/etc/guest-session/skel/
  if [ -d "$gs_skel" ] && [ -n "`find $gs_skel -type f`" ]; then
    cp -rT $gs_skel "$HOME"
  else
    cp -rT /etc/skel/ "$HOME"
  fi
  chown -R $USER:$USER "$HOME"
  usermod -d "$HOME" "$USER"

  # Load restricted session
  #dmrc='[Desktop]\nSession=guest-restricted'
  #/bin/echo -e "$dmrc" > "$HOME"/.dmrc

  chown -R $USER:$USER "$HOME"

  echo $USER  
}

remove_account ()
{
  USER=$1
  
  PWENT=`getent passwd "$USER"` || {
    echo "Error: invalid user $USER"
    exit 1
  }
  USERUID=`echo "$PWENT" | cut -f3 -d:`
  HOME=`echo "$PWENT" | cut -f6 -d:`

  # deluser is provided by the adduser package on Debian/Ubuntu. useradd
  # doesn't have a '--system' parameter, which causes deluser to only delete
  # system users, so this will be handled using this script.
  SYS_UID_MIN="$(cat /etc/login.defs | grep SYS_UID_MIN | awk '{print $2}')"
  SYS_UID_MAX="$(cat /etc/login.defs | grep SYS_UID_MAX | awk '{print $2}')"
  if [ "$USERUID" -lt "$SYS_UID_MIN" ] || [ "$USERUID" -gt "$SYS_UID_MAX" ]; then
    echo "Error: user $USER is not a system user."
    exit 1
  fi

  if [ "${HOME}" = "${HOME#/tmp/}" ]; then
    echo "Error: home directory $HOME is not in /tmp/."
    exit 1
  fi

  # kill all remaining processes
  while ps h -u "$USER" >/dev/null; do 
    killall -9 -u "$USER" || true
    sleep 0.2; 
  done

  umount "$HOME" || umount -l "$HOME" || true
  rm -rf "$HOME"

  # remove leftovers in /tmp
  find /tmp -mindepth 1 -maxdepth 1 -uid "$USERUID" -print0 | xargs -0 rm -rf || true

  #deluser --system "$USER"
  userdel "$USER"
}

case "$1" in
  add)
    add_account
    ;;
  remove)
    if [ -z $2 ] ; then
      echo "Usage: $0 remove [account]"
      exit 1
    fi
    remove_account $2
    ;;
  *)
    echo "Usage: $0 add|remove"
    exit 1
esac
```

AdnanHodzic, 2023-09-30:
With [release of v2.0.0](https://github.com/AdnanHodzic/auto-cpufreq/releases/tag/v2.0.0) now auto-cpufreq has GUI. Hence, tray can be replicated in [GNOME by adding auto-cpufreq to "dash" by right clicking and selecting "pin".](https://youtu.be/SPGpkZ0AZVU?si=SaJ-jcFCJ99uEph2&t=420)

Closing the issue.

closed 2023-09-30 by AdnanHodzic