AdnanHodzic/auto-cpufreq #91: AUR installer issues

ntropy83, 2020-08-10:
Hi there, thanks for the latest update.
However after updating it, my CPU Governor suddenly got stuck on performance with AC plugged in and couldnt be reverted. So I purged the software and installed again. If I run auto-cpufreq-install tho the installation fails with a copy error with a wrong path to the auto-cpufreq.service file. 

Missing file in /usr/local/share/auto-cpufreq/scripts

If I copy it manually there, it works, but then the service fails to start on a missing file: 

```
auto-cpufreq.service: Failed at step EXEC spawning /usr/local/bin/auto-cpufreq
```

Seems like some linking errors.

If I install via " auto-cpufreq --install " setup will break not finding the auto-cpufreq-install.sh nor the auto-cpufreq-remove.sh in /usr/local/auto-cpufreq. 

Putting it all manually where it belongs, does work, yet on AC plugged in the CPU governor is now always on performance for me and only scales back on battery.

crian, 2020-08-10:
Don't run the install command after you installed the AUR package.
Just enable the systemd service.

AdnanHodzic, 2020-08-12:
[README ](https://github.com/AdnanHodzic/auto-cpufreq)updated, I hope this clarifies any confusion. Please submit a new PR if you think otherwise.

ntropy83, 2020-08-13:
All good that worked out. Meanwhile I dug a bit into the code and know more now how it is working. Python is pretty new to me but intresting. :)

I have modified the code for me, so the cpu stays on powersave with AC plugged in as well. This is better for this hell of a Coffe Lake Refresh ++ intel of mine that runs pretty hot real quick. :)

One last thing for the Readme: systemctl enable auto-cpufreq would make it availiable between restarts and symlink it to systemd. The start command does only launch it for the current session.

AdnanHodzic, 2020-08-14:
> All good that worked out. Meanwhile I dug a bit into the code and know more now how it is working. Python is pretty new to me but intresting. :)

:+1: 

> I have modified the code for me, so the cpu stays on powersave with AC plugged in as well. This is better for this hell of a Coffee Lake Refresh ++ intel of mine that runs pretty hot real quick. :)

Soon enough a settings file will allow you to do this kind of modifications, once work in #26 is done. 

> One last thing for the Readme: systemctl enable auto-cpufreq would make it availiable between restarts and symlink it to systemd. The start command does only launch it for the current session.

Didn't think it was necessary to mention that to Arch users :wink: Regardless, I've updated [Arch section of README once again](https://github.com/AdnanHodzic/auto-cpufreq#aur-package-archmanjaro-linux) and pointed what could be permanent solution to this problem in: #96

drws, 2020-12-13:
The question how to properly clean up if `auto-cpufreq --install` was accidentially ran is remaining. If Arch is kept as an exampe, is it enough to remove the package and install it properly again or is there anything else that needs to be done?

AdnanHodzic, 2020-12-15:
You accidentally ran `auto-cpufreq --install`? Run `auto-cpufreq --remove` and it's gone.

Also removing the package should remove both auto-cpufreq and running install.

drws, 2020-12-18:
I was going through the docs but missed it somehow. Thank you for clearing it up!