AdnanHodzic/auto-cpufreq #98: Problem enabling Turbo on Ryzen CPUS

rrodriguez81, 2020-08-29:
Hi,

I own an AMD Ryzen 3700U CPU, the turbo isn't enabled when one thread reach 100% of CPU usage. I need at least two of them to reach the 25% of CPU Load.

I fixed it modifying the source code a bit. Since I can create a new branch and push the possible change, I'm attaching the modified file, maybe it can be useful to other users,

Thanks for you good work,
[core.zip](https://github.com/AdnanHodzic/auto-cpufreq/files/5145595/core.zip)

AdnanHodzic, 2020-08-29:
Great! 

Could you please create a [PR from a fork](https://docs.github.com/en/github/collaborating-with-issues-and-pull-requests/creating-a-pull-request-from-a-fork)? This would help me in review and also to give you give credit for your contribution, thanks.

rrodriguez81, 2020-08-29:
Done! take a look at it. maybe is an easy but good solution.

Best regards.

AdnanHodzic, 2020-08-29:
Great, thank you. Let me look into it and test couple of things and I'll get back to you.

AdnanHodzic, 2020-08-31:
Fixed as part of #99

closed 2020-08-31 by AdnanHodzic