AdnanHodzic/auto-cpufreq #100: GUI for CPU freq

yasharya2901, 2020-08-31:
###Feature Request
As auto-cpufreq is a command line, having a GUI for it would be so much awesome as GUI is just awesome, isn't it? Plus, it can be less intimidating for a new user to navigate through GUI rather than CLI.

AdnanHodzic, 2020-08-31:
I believe this is a duplicate of: #87

AdnanHodzic, 2021-10-16:
I'll close this issue and let's track its status in #87. After  [v1.7.0 release](https://github.com/AdnanHodzic/auto-cpufreq/releases/tag/v1.7.0) release and config/settings file going live. This feature probably next on my priority list. If you want to give it a try at implementing this feature, I'll make sure to give you credit as part of [releases page](https://github.com/AdnanHodzic/auto-cpufreq/releases).

closed 2021-10-16 by AdnanHodzic

AdnanHodzic, 2023-09-30:
With [release of v2.0.0](https://github.com/AdnanHodzic/auto-cpufreq/releases/tag/v2.0.0) now auto-cpufreq has GUI. Hence, tray can be replicated in [GNOME by adding auto-cpufreq to "dash" by right clicking and selecting "pin".](https://youtu.be/SPGpkZ0AZVU?si=SaJ-jcFCJ99uEph2&t=420)