AdnanHodzic/auto-cpufreq #104: No full performance when running benchmarks

Ritte88, 2020-09-02:
I'm running Arch with the latest kernel. I tried to run som benchmarks with ryzedadj (set to low tdp) and I forgot that I had auto-cpufreq enabled when I run the benchs. But in my mind it should not matter since auto-cpufreq allows full performance when needed.
Anyhow, here's geekbench with the only difference being auto-cpufreq:
https://i.imgur.com/3AGytJr.jpg
https://i.imgur.com/wCW0uBK.jpg

Is it expected that auto-cpufreq limits the performance, or is this a bug?

AdnanHodzic, 2020-09-02:
> It should not matter since auto-cpufreq allows full performance when needed.

That's correct.

But I don't understand what's the problem. Are you saying you're getting lower performance while auto-cpufreq is enabled? Which one of these are you getting with auto-cpufreq running? https://i.imgur.com/wCW0uBK.jpg or https://i.imgur.com/wCW0uBK.jpg

Ritte88, 2020-09-02:
The image with lower single-core is auto-cpufreq. 
The problem is that this is not expected behavior, if auto-cpufreq allows for full performance (if needed). 

The singe core,performance should be the same?

I found this about by forgetting to disable auto-coufreq, and the finding made me curious if this is expected behavior.

AdnanHodzic, 2020-09-02:
Right, which version of auto-cpufreq are you using? 

@crian are AUR package changes made automatically based on git changes, or you have to trigger then each time based on release?

Because as part of 1.2.2 release some changes were made to address single core performance: #99 | @Red-Eyed @rrodriguez81 could this be related to that single core turbo trigger delay?

rrodriguez81, 2020-09-02:
@AdnanHodzic is posible that in a benchmark, the delay of 3-4s of triggering turbo can be very bad for benchmarks.

crian, 2020-09-02:
The AUR package is build from the latest git commit.

rrodriguez81, 2020-09-02:
@crian the versión os 1.2.1 or 1.2.2?

crian, 2020-09-02:
> @crian the versión os 1.2.1 or 1.2.2?

If you build it now the pkg will be based on commit 4c2670cae52bc7118450dc90f9c9f7a3e3568761

Ritte88, 2020-09-02:
I built the pkg about 12h ago.

AdnanHodzic, 2020-09-03:
@Ritte88 this problem should be fixed in #105 

Not sure how long it takes for AUR package build to be triggered. But please verify this problem is gone once you're running latest changes.

closed 2020-09-03 by AdnanHodzic