AdnanHodzic/auto-cpufreq #107: If you could add support for Solus that would be swell.

Philbohackinz, 2020-09-06:
Distribution: Solus
Release: 4.1
Codename: fortitude

AdnanHodzic, 2020-09-06:
Can you tell me what's the problem you get after installing the snap, and what's the error you get during the installation of [auto-cpufreq-installer](https://github.com/AdnanHodzic/auto-cpufreq#auto-cpufreq-installer)?

natri23, 2020-09-23:
> **$ - sudo ./auto-cpufreq-installer**
> Password:
> 
> -------------------------- auto-cpufreq installer -----------------------------
> 
> Welcome to auto-cpufreq tool installer.
> 
> Options:
> 
> [I]nstall
> [R]emove
> [Q]uit
> 
> Select a key: [i/r/q]: i
> 
> -------------------------------------------------------------------------------
> 
> Didn't detect Debian or RedHat based distro.
> 
> To complete installation, you need to:
> Install: python3, pip3 and inxi
> 
> Install necessary Python packages:
> pip3 install psutil click distro power
> 
> Run following sequence of lines:
> 
> -----
> 
> python3 setup.py install --record files.txt
> mkdir -p /usr/local/share/auto-cpufreq/
> cp -r scripts/ /usr/local/share/auto-cpufreq/
> 
> -----
> 
> After which tool is installed, for full list of options run:
> auto-cpufreq
> 
> -------------------------------------------------------------------------------
> 
> Consider creating a feature request to add support for your distro:
> https://github.com/AdnanHodzic/auto-cpufreq/issues/new
> 
> Make sure to include following information:
> 
> Distribution: Solus
> Release: 4.1
> Codename: fortitude
> 
> -------------------------------------------------------------------------------
> **$ -  python3 setup.py install --record files.txt**
> running install
> running bdist_egg
> running egg_info
> error: [Errno 13] Permission denied

I can force installed it with:
> sudo python3 setup.py install --record files.txt

After that I edited /usr/bin/auto-cpufreq/scripts/auto-cpufreq.service and replace 
> ExecStart=/usr/local/bin/auto-cpufreq --daemon

with

> ExecStart=/usr/bin/auto-cpufreq --daemon

then it would work with --intstall

AdnanHodzic, 2020-09-23:
@natri23 I would encourage you to create a [PR request](https://docs.github.com/en/github/collaborating-with-issues-and-pull-requests/creating-a-pull-request) with changes to installer which work for you.

natri23, 2020-09-24:
#122 Added simple instruction lines for Solus

AdnanHodzic, 2020-09-24:
Support for Solus have been added as part of #122, changes are live as [part of 1.3.3 release](https://github.com/AdnanHodzic/auto-cpufreq/releases/tag/v1.3.3). 

Closing this issue, if the problem persists please feel free to re-open it.

closed 2020-09-24 by AdnanHodzic