AdnanHodzic/auto-cpufreq #118: Turbo boost enabled under low load

luispabon, 2020-09-17:
### Error output:
```text
n/a
```
---

### System information:
```text
-------------------------------------------------------------------------------

Linux distro: Ubuntu 20.04.1 LTS (Focal Fossa)
Linux kernel: 5.4.0-47-generic

Snap package: yes

Python: 3.8.2
psutil package: 5.7.2
platform package: 1.0.8
click package: 7.1.2
distro package 1.5.0

Battery is: charging

auto-cpufreq system resource consumption:
cpu usage: 0.0 %
memory use: 0.05 %

Processor: Intel(R) Core(TM) i7-7700HQ CPU @ 2.80GHz
Cores: 8
Architecture: x86_64
Driver: intel_pstate

------------------------------ Current CPU states ------------------------------

CPU max frequency: 3800 MHz
CPU min frequency: 800 MHz

CPU frequency for each core:

CPU0: 3635 MHz
CPU1: 3552 MHz
CPU2: 3684 MHz
CPU3: 3554 MHz
CPU4: 3529 MHz
CPU5: 3542 MHz
CPU6: 3713 MHz
CPU7: 3578 MHz

CPU usage per each core:

CPU0: 1.0 %
CPU1: 1.0 %
CPU2: 1.0 %
CPU3: 0.0 %
CPU4: 0.0 %
CPU5: 0.0 %
CPU6: 1.0 %
CPU7: 1.0 %

Temperature for each physical core:

CPU0 temp: 49°C
CPU1 temp: 48°C
CPU2 temp: 50°C
CPU3 temp: 47°C

Total CPU usage: 0.6 %
Total system load: 0.85 

Currently using: performance governor
Currently turbo boost is: on

-------------------------------------------------------------------------------
```
---

AdnanHodzic, 2020-09-17:
That's for confirming this. As [I mentioned](https://github.com/AdnanHodzic/auto-cpufreq/issues/109#issuecomment-691314410) in #109  currently `auto-cpufreq` is very "turbo mode" trigger friendly when charging. 

I've marked this big as duplicate, and if possible let's continue this discussion in #109.

AdnanHodzic, 2020-09-17:
@luispabon I made some [changes in turbo-lover](https://github.com/AdnanHodzic/auto-cpufreq/tree/turbo-lover) branch, you can get them by installing `auto-cpufreq` from snap beta channel:

```
sudo snap install auto-cpufreq --beta
```

Now it's using more conservative settings and turbo isn't triggered as easily as before. Let me know how it works for you.

luispabon, 2020-09-17:
Copy that, I will do so as soon as I'm back to charging tomorrow :+1:

luispabon, 2020-09-18:
@AdnanHodzic definitely much better than before.

luispabon, 2020-09-18:
Do you want to keep this open until the change is released to non-beta? Or should I just close it?

AdnanHodzic, 2020-09-18:
> @AdnanHodzic definitely much better than before.

Good to hear, I could make turbo trigger even more conservative, but this seems like the right balance. 

> Do you want to keep this open until the change is released to non-beta? Or should I just close it?

Don't worry about it, once I've released 1.3.2 final and promoted build to snap stable channel, I'll close it myself. I want to test it more myself and for @marc0der  to confirm the same in #109

dnisyd, 2020-09-19:
How about making it configurable which lets us use the default turbo boost load balance trigger or use a self defined threshold?

AdnanHodzic, 2020-09-19:
@dnisyd while I was working on this I was thinking about the same thing. This would be ideal candidate for #26, but I'm not sure when I'll have time to start working on it. 

Also, until #26 is implemented, we need to have good "defaults" for triggering turbo. What was there before was too "trigger happy" in terms of turning on turbo.

AdnanHodzic, 2020-09-22:
After thoroughly testing changes for few days, I've merged them with `master` branch and made [1.3.2 release](https://github.com/AdnanHodzic/auto-cpufreq/releases/tag/v1.3.2). These changes have also been promoted to snap `stable` channel. 

If you've been using beta channel you can switch to stable by running:

```
sudo snap switch --channel=stable auto-cpufreq
```

closed 2020-09-22 by AdnanHodzic