AdnanHodzic/auto-cpufreq #121: sudo auto-cpufreq

AceOfSpadez54, 2020-09-20:
### Error output:

Installing necessary Python packages

ModuleNotFoundError: No module named 'setuptools'

installing auto-cpufreq tool
-----------------------------------------------------
Traceback (most recent call last):
  File "setup.py", line 3, in <module>
    from setuptools import setup
/usr/bin/python3: No module named pip
------------------------------------------------------
sudo: auto-cpufreq: command not found

### System information:

sudo: auto-cpufreq: command not found
Further context:
Using LMDE 4 "Debbie". Two missing module errors as pasted above. Despite those errors, installer script says the package installed successfully. However, when I try to run "sudo auto-cpufreq" I get a command not found error, also pasted above.

AdnanHodzic, 2020-09-20:
I just made changes which should fix your problem, pull the latest changes and run `auto-cpufreq-installer --installer` again.

AdnanHodzic, 2020-09-22:
Closing due to inactivity.

closed 2020-09-22 by AdnanHodzic