AdnanHodzic/auto-cpufreq #130: [request] NixOS support

Creator54, 2020-11-10:
Please add support for NixOS

Technical27, 2020-12-08:
I'm working on a package, but I get this error: #138. Other than that it appears to be mostly functional.

AdnanHodzic, 2020-12-08:
@Technical27 great, looking forward to a PR from you soon.

AdnanHodzic, 2021-02-04:
@Technical27 was going through open issues, are there any updates on this?

Technical27, 2021-02-04:
The package is now in nixos-unstable, but the module isn't. So instead of `services.auto-cpufreq.enable = true`
you need to put
```
environment.systemPackages = [ pkgs.auto-cpufreq ];
systemd.packages = [ ckgs.auto-cpufreq ];
systemd.services.auto-cpufreq.path = with pkgs; [ bash coreutils ];
```
for now. Someone should be reviewing the module now, that hopefully should get merged soon. When that gets merged, I'll close this.

AdnanHodzic, 2021-02-04:
Nice, glad to hear this! Please keep me updated on progress of this, as in the end it would be nice to have instractions on how to install auto-cpufreq on NixOS as part of [ReadMe](https://github.com/AdnanHodzic/auto-cpufreq/blob/master/README.md).

Technical27, 2021-02-10:
More news, the module has been merged, and is now in NixOS unstable. A PR to update the package to 1.6.1 is also open, and I will review that soon.

AdnanHodzic, 2021-02-11:
> More news, the module has been merged, and is now in NixOS unstable. A PR to update the package to 1.6.1 is also open, and I will review that soon.

Great news, keep them coming!

AdnanHodzic, 2021-10-16:
Are there any updates on this?