AdnanHodzic/auto-cpufreq #132: Watch CPU frequency in near real time.

joaoherberto, 2020-12-03:
This is not an issue, I just want to share this code snippet for monitoring CPU frequency in almost realtime. I use auto-cpufreq on all my Linux devices, even desktops and it is always nice to see if it is working properly or not. Note: I found this code months ago and unfortunately I don't know whom to give credit. 

`watch -n.1 "cat /proc/cpuinfo | grep \"^[c]pu MHz\""`

AdnanHodzic, 2020-12-06:
> I use auto-cpufreq on all my Linux devices, even desktops and it is always nice to see if it is working properly or not

Glad to hear this!

In that case I'm marking this as a [request]. I think this feature would be perfectly implemented as part of what's requested in #87 

I don't know when I'll have time to implement this feature, so if you're up and know bit of Python please make the changes yourself and submit a PR. I'll be glad to do the code review. 

If anything you'll be given credit as [part of Releases](https://github.com/AdnanHodzic/auto-cpufreq/releases) :)

netikras, 2021-12-20:
Two things to consider:
1. cat + grep -- that's unnecessary. Especially when grep can easily do what cat does -- read a file. Just do `grep <pattern> <file>`
2. Current refresh rate is what, 4 seconds? Tuning it down to 0.1 second would mean that auto-cpufreq will utilize 40x more CPU resources. This makes it possible for auto-cpufreq itself to cause high CPU usage and switch to Performance mode, meaning the governor will always be at Performance, even when the computer is idling. This risk is especially worrying in smaller, older computers, esp. single-core ones w/o HT.

AdnanHodzic, 2021-12-20:
Tat's an excellent point @netikras and same conclusion I came to when I was thinking about this last time. Definitely something to keep on mind if we go forward with implementation of one this feature request.

AdnanHodzic, 2022-06-06:
In [latest release 1.9.4,](https://github.com/AdnanHodzic/auto-cpufreq/releases/tag/v1.9.4) changes have been made where auto-cpufreq will refresh frequencies in (near) real time and counter has been removed.

Refresh rate is still around every 3-4 seconds, as through my analysis and testing anything below this timeframe will only cause tool to use unnecessary amount of resources, which would defeat purpose of this tool.

With that said I'll close this feature request.

closed 2022-06-06 by AdnanHodzic