AdnanHodzic/auto-cpufreq #137: Display auto-cpufreq version and update availability

alkaphreak, 2020-12-08:
It could be usefull to display current version and update availabilty.

AdnanHodzic, 2020-12-08:
Yes agreed, there are some attempts to detect version with `sudo auto-cpufreq --debug`, but it's not fully unified as some people install it using `auto-cpufreq-installer `and others using snap or AUR package. But I was planning to do this regardless. 

Regarding update availability, you can simply click on ["Watch" notifications](https://docs.github.com/en/enterprise/2.15/user/articles/watching-and-unwatching-repositories) and can be notified about all changes or only "Release" changes.

alkaphreak, 2020-12-09:
Did a previous install with snap version but i was wondering if it kept it to up to date so i needed to check the installed version ... no success.

AdnanHodzic, 2020-12-09:
> Did a previous install with snap version but i was wondering if it kept it to up to date so i needed to check the installed version ... no success.

You can check this by running:

```
snap list | grep auto-cpufreq
```

alkaphreak, 2020-12-10:
Nice, thanks for the tips

AdnanHodzic, 2021-02-06:
Changes were made in [1.6.0 release](https://github.com/AdnanHodzic/auto-cpufreq/releases/tag/v1.6.0). Where version that's being used is being showed for Snap/AUR/auto-cpufreq-installre (source).

closed 2021-02-06 by AdnanHodzic