AdnanHodzic/auto-cpufreq #157: Deb package?

Mek101, 2021-01-25:
Would it be possible to release a deb package, instead of using a shell script?

MischievousTomato, 2021-01-26:
I second this, as I'm currently using Debian, this will also benefit Ubuntu users.

k-chapin, 2021-01-30:
I also agree. A deb would be great!

utk-dev, 2021-07-30:
@MischievousTomato I believe that Snap works on Debian as well? (I don't know I don't use Debian). Here on Ubuntu, it just works... no configuration.. no installation steps, just one click and you are rolling.

AdnanHodzic, 2021-10-16:
I suggest to install it using [Snap package](https://github.com/AdnanHodzic/auto-cpufreq#snap-store)? As it runs in a container and isolates its Python env from your env. so you shouldn't experience Python related errors.

It's ironic what I'm saying here since I used to be a Debian package maintainer. I still think it would be cool to have it packaged for Debian and all its derivatives, but I don't have time to do this now. 

If any of you want to take a shot at packing auto-cpufreq, I think your best reference is to follow [Debian Mentors page](https://mentors.debian.net/) and someone will help you along the way. If this happens, the least I can do is to give you credits as part [releases page](https://github.com/AdnanHodzic/auto-cpufreq/releases).

xtronics, 2021-10-17:
Snap is a bad idea outside of testing - when everyone uses the same lib- the libs get better.  This is the same reason Windoze software is so buggy.

Also, realize that Debian is what so many other distributions pull from - if you create a Debian package, more eyes on the code - more people to help. 

So please consider a Debian package.

AdnanHodzic, 2023-10-01:
Update: now there's a [RFP to make auto-cpufreq as part of Debian repositories. ](https://bugs.debian.org/cgi-bin/bugreport.cgi?bug=1053294)

emfox, 2026-09-17:
I take the [Debian RFP](https://bugs.debian.org/cgi-bin/bugreport.cgi?bug=1053294) and will package it soon.

emfox, 2026-09-17:
Hello, I've upload auto-cpufreq package to Debian NEW queue, it'll be in Debian unstable in a few days if everything's ok.

emfox, 2026-09-24:
Hello, version 3.1.0 of auto-cpufreq was just accepted into debian. It's not enabled by default, please read README.Debian.

AdnanHodzic, 2026-10-03:
> Hello, I've upload auto-cpufreq package to Debian NEW queue, it'll be in Debian unstable in a few days if everything's ok.

@emfox these are amazing news! Will close this issue and #950.

Most importantly, PR #951 can also be closed now. Its aim was to create created the auto-cpufreq debian (deb) package but was never merged because it didn't meet quality standards. Let's continue this discussion there.

closed 2026-10-03 by AdnanHodzic

AdnanHodzic, 2026-10-05:
Changes mentioned as part of [auto-cpufreq v3.2.0 release](https://github.com/AdnanHodzic/auto-cpufreq/releases).