AdnanHodzic/auto-cpufreq #169: [Bug] Wrong tag - 1.5.4.r4.gd062ca9

schinfo, 2021-02-04:
You've not set the right tag - was 1.5.5, but now it's 1.5.4 and should be 1.5.6

AdnanHodzic, 2021-02-04:
Sorry, which tag are you talking about? All tags are listed [under releases](https://github.com/AdnanHodzic/auto-cpufreq/releases).

I'm also currently working on new changes and planning to make a new tag and release `1.6.0` in upcoming few days.

schinfo, 2021-02-04:
And where is 1.5.6 ?

AdnanHodzic, 2021-02-04:
It doesn't exist yet, earlier today I worked on a branch called [v1.5.6](https://github.com/AdnanHodzic/auto-cpufreq/tree/v1.5.6) which was merged into master and there's mention of [1.5.6 in snapcraft,yml](https://github.com/AdnanHodzic/auto-cpufreq/blob/master/snap/snapcraft.yaml#L3), as it's been pushed as such to Snap beta channel,but this will most likely be changed to 1.6.0

yochananmarqos, 2021-02-04:
Perhaps @schinfo asked the wrong question. What happened to the 1.5.5 tag? The latest. commit comes out as 1.5.4.r4.gd062ca9.

AdnanHodzic, 2021-02-04:
> Perhaps @schinfo asked the wrong question. What happened to the 1.5.5 tag? The latest commit comes out as 1.5.4.r4.gd062ca9, but there was a 1.5.5.r0.gbc601cf.

I'm not sure ... what could've happened is that once [1.5.5 was released](https://github.com/AdnanHodzic/auto-cpufreq/releases/tag/v1.5.5) based on 3985d3e. As additional reports on chassis-type were coming in [as part of my Tweet,](https://twitter.com/fooctrl/status/1357230198857564160) I updated the list and I amended my last commit and I did a force push.

After that I made few more commits/branches as I'm working on a new tag/release which hasn't been created yet.

AdnanHodzic, 2021-02-06:
Closing with release of [1.6.0 was released](https://github.com/AdnanHodzic/auto-cpufreq/releases/tag/v1.6.0).

closed 2021-02-06 by AdnanHodzic