AdnanHodzic/auto-cpufreq #175: question: which TLP settings are recommended?

bbigras, 2021-02-10:
> Just disable governor and turbo boost settings of TLP, then you’re fine.

source: https://www.reddit.com/r/linux/comments/ejxx9f/github_autocpufreq_automatic_cpu_speed_power/fd4y36k/

Is this what I need for boost?
```
        CPU_BOOST_ON_AC=0
        CPU_BOOST_ON_BAT=0
```

For the governor I'm not sure. In the doc https://linrunner.de/tlp/settings/processor.html , I see how to set a specific governor but not how to tell TLP to leave the governor alone.

AdnanHodzic, 2021-02-11:
`auto-cpufreq` has all the necessary functionality to do this on your behalf automatically. Hence, I will not recommend settings you should use as part of TLP.

I would also suggest to move this question to [auto-cpufreq Discussions](https://github.com/AdnanHodzic/auto-cpufreq/discussions).

closed 2021-02-11 by AdnanHodzic

Silejonu, 2021-09-04:
@AdnanHodzic The ReadMe still shows:

> Please note: this tool doesn't conflict and [works great in tandem with TLP](https://www.reddit.com/r/linux/comments/ejxx9f/github_autocpufreq_automatic_cpu_speed_power/fd4y36k/).

I understood it as "you should install TLP in tandem with auto-cpufreq, and it will automatically configure TLP to not conflict with cpufreq". I suppose this could use an update, correct?