AdnanHodzic/auto-cpufreq #180: (Fedora33) cupfreq-installer seems to install all but command auto-cpufreq returns "command not found"

Shogu, 2021-02-13:
### Error output:
```text
bash: auto-cpufreq: command not found...
```
---

### System information:
```text
bash: auto-cpufreq: command not found...
```
---

Shogu, 2021-02-13:
Hi!

On Fedora, using installer and git source, everything seems to be installed correctly if i trust terminal, but commands fails 
I read same issue with snap package on Fedora.

Any idea? Thanks a lot!

Ugo - France

AdnanHodzic, 2021-02-14:
Hi Ugo,

Yes, this is a known issue on Fedora & Snap package, please see second paragraph of "Please note" section for Fedora users as part of [Installing auto-cpufreq from Snap store](https://github.com/AdnanHodzic/auto-cpufreq#snap-store).

Since it's a known issue and there's no permanent solution for it now, I'll close the issue, feel free to re-open it if this doesn't solve your problem.

Shogu, 2021-02-14:
Thanks Adnan.

But on l'y Fedora this trouble appears with installer script, not with Snap.

AdnanHodzic, 2021-02-14:
You install auto-cpufreq with auto-cpufreq-installer and you run it and you get `bash: auto-cpufreq: command not found...`?

Can you provide `auto-cpufreq --debug` output?

Shogu, 2021-02-14:
Thanks a lot for spending time for me - for us with your tool!

I tried again after yesterday fails and do not know why, but i used same commands and today everything runs fine.

Sorry for the false alert.

Greetings from France!

Shogu

closed 2021-02-14 by AdnanHodzic