AdnanHodzic/auto-cpufreq #191: Unnecessary disk io with custom cpufreqctl

theol0403, 2021-03-15:
Upon doing some profiling on my computer to reduce ssd wear, I noticed that this tool continuously writes to `/usr/bin/cpufreqctl.auto-cpufreq` every refresh cycle. This seems unnecessary, and I'm a little unsure about the negative impacts of this.

Would it be possible to only install the custom script one time, as part of the setup process, and then leave it there? I'm not sure why it's not, could this be a bug?

Many thanks!

AdnanHodzic, 2021-03-20:
Is this is the still the case [1.6.4](https://github.com/AdnanHodzic/auto-cpufreq/releases/tag/v1.6.4)? 

@marc0der considering your involvement with this one, any thoughts?

marc0der, 2021-03-20:
@AdnanHodzic the work I did only introduced a parallel version of this executable but did nothing in terms of how it is used or placed on the file system. I can take a look to see if this can be optimised further.

AdnanHodzic, 2021-03-20:
@marc0der I know, please don't take this the wrong way :) 

I noticed increase in issues related to `cpufreqctl.auto-cpufreq` (although I don't see any reason for it). Hence I'm trying to eliminate false positives and attempting to figure out what the real cause is.

@theol0403 please confirm if this is still the problem with 1.6.4

theol0403, 2021-03-20:
At the time of reporting, I was installing from the latest git commit, so I think so.

AdnanHodzic, 2021-03-20:
> At the time of reporting, I was installing from the latest git commit, so I think so.

Could you please include output of:
```
sudo auto-cpufreq --debug
```

theol0403, 2021-03-20:
```
✗ sudo auto-cpufreq --debug

-------------------------------------------------------------------------------

Linux distro: Fedora 34
Linux kernel: 5.11.7-300.fc34.x86_64
Processor: Intel(R) Core(TM) i7-8550U CPU @ 1.80GHz
Cores: 8
Architecture: x86_64
Driver: intel_pstate

------------------------------ Current CPU stats ------------------------------

CPU max frequency: 1800 MHz
CPU min frequency: 400 MHz

Core    Usage   Temperature     Frequency
CPU0:    28.0%     57 °C     1800 MHz
CPU1:     4.0%     58 °C     2000 MHz
CPU2:     4.0%     55 °C     2000 MHz
CPU3:    10.1%     57 °C     2000 MHz
CPU4:     5.1%     57 °C     2000 MHz
CPU5:    15.0%     58 °C     2000 MHz
CPU6:     2.0%     55 °C     2000 MHz
CPU7:     8.9%     57 °C     2000 MHz

auto-cpufreq version:
Git commit: e1409aa

Python: 3.9.2
psutil package: 5.8.0
platform package: 1.0.8
click package: 7.1.2
distro package 1.5.0

Computer type: Convertible
Battery is: charging

auto-cpufreq system resource consumption:
cpu usage: 0.0 %
memory use: 0.07 %

Total CPU usage: 14.0 %
Total system load: 1.43
Average temp. of all cores: 56.75 °C

Currently using: performance governor
Currently turbo boost is: off

-------------------------------------------------------------------------------
```

AdnanHodzic, 2021-08-14:
Added [Troubleshooting section to README on how to change CPU management driver](https://github.com/AdnanHodzic/auto-cpufreq#troubleshooting).

Could you please give this a try as I believe it could potentially resolve your problem. If it doesn't please feel free to re-open this issue.

closed 2021-08-14 by AdnanHodzic