AdnanHodzic/auto-cpufreq #192: Cannot deploy on Raspberry Pi 4B with Ubuntu 20.04

Acsigen, 2021-03-27:
### Error output:
```
Traceback (most recent call last):
  File "/usr/local/bin/auto-cpufreq", line 4, in <module>
    __import__('pkg_resources').run_script('auto-cpufreq==1.0', 'auto-cpufreq')
  File "/usr/lib/python3/dist-packages/pkg_resources/__init__.py", line 667, in run_script
    self.require(requires)[0].run_script(script_name, ns)
  File "/usr/lib/python3/dist-packages/pkg_resources/__init__.py", line 1470, in run_script
    exec(script_code, namespace, namespace)
  File "/usr/local/lib/python3.8/dist-packages/auto_cpufreq-1.0-py3.8.egg/EGG-INFO/scripts/auto-cpufreq", line 151, in <module>
  File "/usr/lib/python3/dist-packages/click/core.py", line 764, in __call__
    return self.main(*args, **kwargs)
  File "/usr/lib/python3/dist-packages/click/core.py", line 717, in main
    rv = self.invoke(ctx)
  File "/usr/lib/python3/dist-packages/click/core.py", line 956, in invoke
    return ctx.invoke(self.callback, **ctx.params)
  File "/usr/lib/python3/dist-packages/click/core.py", line 555, in invoke
    return callback(*args, **kwargs)
  File "/usr/local/lib/python3.8/dist-packages/auto_cpufreq-1.0-py3.8.egg/EGG-INFO/scripts/auto-cpufreq", line 97, in main
  File "/usr/local/lib/python3.8/dist-packages/auto_cpufreq-1.0-py3.8.egg/auto_cpufreq/core.py", line 917, in sysinfo
IndexError: list index out of range
```
---

### System information:
```
Linux distro: Ubuntu 20.04 focal
Linux kernel: 5.4.0-1032-raspi
Processor:
Cores: 4
Architecture: aarch64
Driver: cpufreq-dt
```
---

AdnanHodzic, 2021-08-14:
How have you installed auto-cpufreq? If you installed it using [auto-cpufreq-installer](https://github.com/AdnanHodzic/auto-cpufreq/#auto-cpufreq-installer) have you tried installing it using [Snap package](https://github.com/AdnanHodzic/auto-cpufreq/#snap-store)?

Acsigen, 2021-08-18:
I installed it using auto-cpufreq-installer. I did not try to install it using the snap package because I do not use snap packages.

Dakad, 2021-09-18:
@Acsigen  I got the same error message and I just forgot to run it with `sudo`

Acsigen, 2021-09-20:
@Dakad I ran it as root at the time

4bitFox, 2021-09-21:
Same exact problem here on Raspberry Pi OS / Raspian:

```
-------------------------------------------------------------------------------

Linux distro: Raspbian GNU/Linux 10 buster
Linux kernel: 5.10.52-v7l+
Processor: ARMv7 Processor rev 3 (v7l)
Cores: 4
Architecture: armv7l
Driver: cpufreq-dt

------------------------------ Current CPU stats ------------------------------

CPU max frequency: 1500 MHz
CPU min frequency: 600 MHz

Traceback (most recent call last):
  File "/usr/local/bin/auto-cpufreq", line 4, in <module>
    __import__('pkg_resources').run_script('auto-cpufreq==1.0', 'auto-cpufreq')
  File "/usr/lib/python3/dist-packages/pkg_resources/__init__.py", line 666, in run_script
    self.require(requires)[0].run_script(script_name, ns)
  File "/usr/lib/python3/dist-packages/pkg_resources/__init__.py", line 1453, in run_script
    exec(script_code, namespace, namespace)
  File "/usr/local/lib/python3.7/dist-packages/auto_cpufreq-1.0-py3.7.egg/EGG-INFO/scripts/auto-cpufreq", line 151, in <module>
  File "/usr/lib/python3/dist-packages/click/core.py", line 764, in __call__
    return self.main(*args, **kwargs)
  File "/usr/lib/python3/dist-packages/click/core.py", line 717, in main
    rv = self.invoke(ctx)
  File "/usr/lib/python3/dist-packages/click/core.py", line 956, in invoke
    return ctx.invoke(self.callback, **ctx.params)
  File "/usr/lib/python3/dist-packages/click/core.py", line 555, in invoke
    return callback(*args, **kwargs)
  File "/usr/local/lib/python3.7/dist-packages/auto_cpufreq-1.0-py3.7.egg/EGG-INFO/scripts/auto-cpufreq", line 71, in main
  File "/usr/local/lib/python3.7/dist-packages/auto_cpufreq-1.0-py3.7.egg/auto_cpufreq/core.py", line 917, in sysinfo
IndexError: list index out of range
```

Acsigen, 2021-10-18:
I will set the problem as solved since it seems that the problem occurs only with autoinstaller on arm