AdnanHodzic/auto-cpufreq #195: Auto-cpufreq wanna burn my pc or what?

arnav7633, 2021-05-28:
Hey there, I am rocking a 2.3ghz amd A8-7410 i had my laptop on charging and the cpu's temp is 96 DEGREE CELSIUS but auto cpu freq is still putting it on performance mode when it should be trying to cool down my pc, Is there something i have to enable to make this happen?

AdnanHodzic, 2021-05-30:
I suggest you take it down a notch. As this (`Auto-cpufreq wanna burn my pc or what?`) is not a way to go about reporting bugs in professional environment and expect someone to assist you with your problem.

In meantime, could you include output of: `sudo auto-cpufreq --debug` and is there any difference in temperature if you plug the laptop out of power?

arnav7633, 2021-05-31:
Happy now?

AdnanHodzic, 2021-05-31:
Reason why it's working so hard is that `Total CPU usage: 67%` which is pretty high. But if computer is in semi-idle and not doing that much temperature can go below 40°. If you actively monitor the state with `sudo auto-cpufreq --stats`, and CPU is running this high, can you check what's process that's doing this?

Also what happens if you plug the laptop out of power, does this lower the temperature? I'm curious to find out as this should force it to use `powersave` governor instead of `performance`.

arnav7633, 2021-05-31:
> Reason why it's working so hard is that `Total CPU usage: 67%` which is pretty high. But if computer is in semi-idle and not doing that much temperature can go below 40°. If you actively monitor the [...]

I actually cant plug out my pc because my batter health is like 12% so if i unplug it dies within a min

AdnanHodzic, 2021-08-14:
Added [Troubleshooting section to README on how to change CPU management driver](https://github.com/AdnanHodzic/auto-cpufreq#troubleshooting).

Could you please give this a try as I believe it will resolve your problem. If it doesn't please feel free to re-open this issue.

closed 2021-08-14 by AdnanHodzic