AdnanHodzic/auto-cpufreq #197: auto-cpufreq folder installed to Home folder

michaelraiwet, 2021-05-31:
Apologies, first time Github comment, and newbie Linux user. I recently installed auto-cpufreq from the auto-cpufreq-installer from the readme. I noticed that it created an auto-cpufreq folder in my Home folder. Is there anyway to remove/move the folder without losing functionality to auto-cpufreq? 

BTW I am on Pop_OS! 20.10 on a Thinkpad T490 if that provides more context.

AdnanHodzic, 2021-05-31:
Yes, uninstall auto-cpufreq, move the directory where you want it to reside permanently and then perform install from there. That's if you're using `auto-cpufreq-installer`, you can also install [using Snap](https://github.com/AdnanHodzic/auto-cpufreq#snap-store) and then you don't have to worry about this.

dropestein, 2021-08-14:
Hi! About this same question: what is the proper way to completely uninstall it?
I tried `sudo auto-cpufreq --remove`, but I still see the `auto-cpufreq` available in my terminal.

AdnanHodzic, 2021-08-14:
> Hi! About this same question: what is the proper way to completely uninstall it?
> I tried `sudo auto-cpufreq --remove`, but I still see the `auto-cpufreq` available in my terminal.

If `sudo auto-cpufreq --remove` is not available, then it means that auto-cpufreq package has not been installed. 

If you installed it using Snap package, you can remove it using: `sudo snap remove auto-cpufreq`

If you installed it using the auto-cpufreq-installer, you can remove it by getting to location where you cloned auto-cpufreq git repo and running: `sudo /auto-cpufreq-installer --uninstall`

closed 2021-08-14 by AdnanHodzic