AdnanHodzic/auto-cpufreq #203: Please help me with the below error as I couldn't use this tool. I am using Zorin OS 15.3

veerabag, 2021-05-31:
### Error output:

Traceback (most recent call last):
  File "/usr/local/bin/auto-cpufreq", line 4, in <module>
    __import__('pkg_resources').run_script('auto-cpufreq==1.0', 'auto-cpufreq')
  File "/usr/lib/python3/dist-packages/pkg_resources/__init__.py", line 658, in run_script
    self.require(requires)[0].run_script(script_name, ns)
  File "/usr/lib/python3/dist-packages/pkg_resources/__init__.py", line 1445, in run_script
    exec(script_code, namespace, namespace)
  File "/usr/local/lib/python3.6/dist-packages/auto_cpufreq-1.0-py3.6.egg/EGG-INFO/scripts/auto-cpufreq", line 25, in <module>
  File "/usr/lib/python3/dist-packages/click/decorators.py", line 170, in decorator
    _param_memo(f, OptionClass(param_decls, **attrs))
  File "/usr/lib/python3/dist-packages/click/core.py", line 1460, in __init__
    Parameter.__init__(self, param_decls, type=type, **attrs)
TypeError: __init__() got an unexpected keyword argument 'hidden'
---

### System information:
```text
Paste here output of the auto-cpufreq --debug
```
---Traceback (most recent call last):
  File "/usr/local/bin/auto-cpufreq", line 4, in <module>
    __import__('pkg_resources').run_script('auto-cpufreq==1.0', 'auto-cpufreq')
  File "/usr/lib/python3/dist-packages/pkg_resources/__init__.py", line 658, in run_script
    self.require(requires)[0].run_script(script_name, ns)
  File "/usr/lib/python3/dist-packages/pkg_resources/__init__.py", line 1445, in run_script
    exec(script_code, namespace, namespace)
  File "/usr/local/lib/python3.6/dist-packages/auto_cpufreq-1.0-py3.6.egg/EGG-INFO/scripts/auto-cpufreq", line 25, in <module>
  File "/usr/lib/python3/dist-packages/click/decorators.py", line 170, in decorator
    _param_memo(f, OptionClass(param_decls, **attrs))
  File "/usr/lib/python3/dist-packages/click/core.py", line 1460, in __init__
    Parameter.__init__(self, param_decls, type=type, **attrs)
TypeError: __init__() got an unexpected keyword argument 'hidden'
[Error.txt](https://github.com/AdnanHodzic/auto-cpufreq/files/6571869/Error.txt)

AdnanHodzic, 2021-05-31:
I guess you installed it using [auto-cpufreq-installer,](https://github.com/AdnanHodzic/auto-cpufreq#auto-cpufreq-installer) what happens if you install it using [Snap package](https://github.com/AdnanHodzic/auto-cpufreq#snap-store)?

veerabag, 2021-05-31:
Thanks for your response.
I got the same error when I install using snap too. 
When I do  "systemctl status snap.auto-cpufreq.service.service" got the below message.

● snap.auto-cpufreq.service.service - Service for snap application auto-cpufreq.
   Loaded: loaded (/etc/systemd/system/snap.auto-cpufreq.service.service; enable
   Active: inactive (dead) since Mon 2021-05-31 14:47:28 EDT; 13min ago
  Process: 1611 ExecStart=/usr/bin/snap run auto-cpufreq.service (code=exited, s
 Main PID: 1611 (code=exited, status=0/SUCCESS)

May 31 14:47:27 veerasZorin systemd[1]: Started Service for snap application aut
lines 1-7/7 (END)...skipping...
● snap.auto-cpufreq.service.service - Service for snap application auto-cpufreq.service
   Loaded: loaded (/etc/systemd/system/snap.auto-cpufreq.service.service; enabled; vendor preset: enabled)
   Active: inactive (dead) since Mon 2021-05-31 14:47:28 EDT; 13min ago
  Process: 1611 ExecStart=/usr/bin/snap run auto-cpufreq.service (code=exited, status=0/SUCCESS)
 Main PID: 1611 (code=exited, status=0/SUCCESS)

May 31 14:47:27 veerasZorin systemd[1]: Started Service for snap application auto-cpufreq.service.

rscm, 2021-06-01:
Duplicate of https://github.com/AdnanHodzic/auto-cpufreq/issues/201

AnetVinci, 2021-06-01:
I am running Zorin as well, and I am facing exact same issue!

AdnanHodzic, 2021-06-01:
But do you get any errors during installation with `auto-cpufreq-installer` or with Snap? Or it goes clean (no errors, warnings) and when you run auto-cpufreq you run into this?

veerabag, 2021-06-01:
No errors at all. It was clean both on CLI & snap.

AdnanHodzic, 2021-06-01:
Verify if it's potentially related to: #196

gageracer, 2021-06-03:
Did you try this? I had the same issue and this fixed it.
`Please note: If you want to install auto-cpufreq daemon, do not run auto-cpufreq --install otherwise you'll run into an issue: #91, #96.

Instead run systemctl start auto-cpufreq to start the service. Run systemctl status auto-cpufreq to see the status of service, and systemctl enable auto-cpufreq for service to persist running accross reboots.`

AdnanHodzic, 2021-06-03:
@veerabag  @AnetVinci can you try what @gageracer suggested?

veerabag, 2021-06-05:
When I run systemctl start auto-cpufreq to start the service as suggested by @gageracer, encountered the below error.

Failed to start auto-cpufreq.service: Unit auto-cpufreq.service not found.

rscm, 2021-06-06:
> @veerabag @AnetVinci can you try what @gageracer suggested?

```
systemctl status auto-cpufreq
Unit auto-cpufreq.service could not be found.
```
They used Arch and installed it through the AUR<br />
Reading the script named `auto-cpufreq-installer`, found the `.service` file under `/usr/local/share/auto-cpufreq/scripts/`

```python
echo -e "\ninstalling auto-cpufreq tool\n"
    install
    update_service_file
    separator
    complete_msg
```

```python
function update_service_file {
  sed -i 's|ExecStart=/usr/local/bin/auto-cpufreq|ExecStart=/usr/bin/auto-cpufreq|' /usr/local/share/auto-cpufreq/scripts/auto-cpufreq.service
}
```

Then I tried creating a softlink to the systemd dir

```Bash
$ sudo ln -s /usr/local/share/auto-cpufreq/scripts/auto-cpufreq.service /etc/systemd/system/auto-cpufreq.service
```

Status after it

```Bash
systemctl status auto-cpufreq
● auto-cpufreq.service - auto-cpufreq - Automatic CPU speed & power optimizer for Linux
   Loaded: loaded (/usr/local/share/auto-cpufreq/scripts/auto-cpufreq.service; linked; vendor preset: enabled)
   Active: inactive (dead)
```

Starting it (trying) and the final status (failure)

```Bash
$ sudo systemctl start auto-cpufreq
```

```Bash
$ systemctl status auto-cpufreq
● auto-cpufreq.service - auto-cpufreq - Automatic CPU speed & power optimizer for Linux
   Loaded: loaded (/usr/local/share/auto-cpufreq/scripts/auto-cpufreq.service; linked; vendor preset: enabled)
   Active: failed (Result: exit-code) since Sat 2021-06-05 23:09:52 -04; 2s ago
  Process: 7140 ExecStart=/usr/local/bin/auto-cpufreq --daemon (code=exited, status=1/FAILURE)
 Main PID: 7140 (code=exited, status=1/FAILURE)

jun 05 23:09:52 legion auto-cpufreq[7140]:   File "/usr/lib/python3/dist-packages/pkg_resources/__init__.py", line 1445, in run_script
jun 05 23:09:52 legion auto-cpufreq[7140]:     exec(script_code, namespace, namespace)
jun 05 23:09:52 legion auto-cpufreq[7140]:   File "/usr/local/lib/python3.6/dist-packages/auto_cpufreq-1.0-py3.6.egg/EGG-INFO/scripts/auto-cpufreq", line 25, in <module>
jun 05 23:09:52 legion auto-cpufreq[7140]:   File "/usr/lib/python3/dist-packages/click/decorators.py", line 170, in decorator
jun 05 23:09:52 legion auto-cpufreq[7140]:     _param_memo(f, OptionClass(param_decls, **attrs))
jun 05 23:09:52 legion auto-cpufreq[7140]:   File "/usr/lib/python3/dist-packages/click/core.py", line 1460, in __init__
jun 05 23:09:52 legion auto-cpufreq[7140]:     Parameter.__init__(self, param_decls, type=type, **attrs)
jun 05 23:09:52 legion auto-cpufreq[7140]: TypeError: __init__() got an unexpected keyword argument 'hidden'
jun 05 23:09:52 legion systemd[1]: auto-cpufreq.service: Main process exited, code=exited, status=1/FAILURE
jun 05 23:09:52 legion systemd[1]: auto-cpufreq.service: Failed with result 'exit-code'.
```

---
And about  #196
```Bash
$ pip3 completion --bash

# pip bash completion start
_pip_completion()
{
    COMPREPLY=( $( COMP_WORDS="${COMP_WORDS[*]}" \
                   COMP_CWORD=$COMP_CWORD \
                   PIP_AUTO_COMPLETE=1 $1 ) )
}
complete -o default -F _pip_completion pip
# pip bash completion end
```

AnetVinci, 2021-06-09:
Anything on this issue yet?

AdnanHodzic, 2021-06-10:
Can you do the following to see if it makes any difference:

Run following in auto-cpufreq dir assuming you first ran and installed [auto-cpufreq using the installer](https://github.com/AdnanHodzic/auto-cpufreq#auto-cpufreq-installer), 
i.e:  `sudo ./auto-cpufreq-installer`

**Option 1:**

```
sudo apt install -y python3-setuptools python3-psutil python3-click python3-distro
````

After this run: `sudo auto-cpufreq --live`

Does it run successfully? If yes, run `sudo auto-cpufreq --remove` and then `sudo auto-cpufreq --install`

**Option 2:**

```
sudo python3 -m pip uninstall -r requirements.txt | # delete all
sudo apt install -y python3-setuptools python3-psutil python3-click python3-distro
```

After this run: `sudo auto-cpufreq --live`

Does it run successfully? If yes, run `sudo auto-cpufreq --remove` and then `sudo auto-cpufreq --install`

rscm, 2021-06-10:
####  Option 1

```
$ sudo apt install -y python3-setuptools python3-psutil python3-click python3-distro
[sudo] contraseña para rscm:            
Leyendo lista de paquetes... Hecho
Creando árbol de dependencias       
Leyendo la información de estado... Hecho
python3-click ya está en su versión más reciente (6.7-3).
fijado python3-click como instalado manualmente.
python3-setuptools ya está en su versión más reciente (39.0.1-2).
Paquetes sugeridos:
  python-psutil-doc
Se instalarán los siguientes paquetes NUEVOS:
  python3-distro python3-psutil
0 actualizados, 2 nuevos se instalarán, 0 para eliminar y 0 no actualizados.
Se necesita descargar 147 kB de archivos.
Se utilizarán 915 kB de espacio de disco adicional después de esta operación.
Des:1 http://cl.archive.ubuntu.com/ubuntu bionic-updates/main amd64 python3-psutil amd64 5.4.2-1ubuntu0.1 [136 kB]
Des:2 http://cl.archive.ubuntu.com/ubuntu bionic/universe amd64 python3-distro all 1.0.1-2 [11,2 kB]
Descargados 147 kB en 2s (94,8 kB/s)       
Seleccionando el paquete python3-psutil previamente no seleccionado.
(Leyendo la base de datos ... 232382 ficheros o directorios instalados actualmente.)
Preparando para desempaquetar .../python3-psutil_5.4.2-1ubuntu0.1_amd64.deb ...
Desempaquetando python3-psutil (5.4.2-1ubuntu0.1) ...
Seleccionando el paquete python3-distro previamente no seleccionado.
Preparando para desempaquetar .../python3-distro_1.0.1-2_all.deb ...
Desempaquetando python3-distro (1.0.1-2) ...
Configurando python3-psutil (5.4.2-1ubuntu0.1) ...
Configurando python3-distro (1.0.1-2) ...
```

```
$ sudo auto-cpufreq --live
Traceback (most recent call last):
  File "/usr/local/bin/auto-cpufreq", line 4, in <module>
    __import__('pkg_resources').run_script('auto-cpufreq==1.0', 'auto-cpufreq')
  File "/usr/lib/python3/dist-packages/pkg_resources/__init__.py", line 658, in run_script
    self.require(requires)[0].run_script(script_name, ns)
  File "/usr/lib/python3/dist-packages/pkg_resources/__init__.py", line 1445, in run_script
    exec(script_code, namespace, namespace)
  File "/usr/local/lib/python3.6/dist-packages/auto_cpufreq-1.0-py3.6.egg/EGG-INFO/scripts/auto-cpufreq", line 25, in <module>
  File "/usr/lib/python3/dist-packages/click/decorators.py", line 170, in decorator
    _param_memo(f, OptionClass(param_decls, **attrs))
  File "/usr/lib/python3/dist-packages/click/core.py", line 1460, in __init__
    Parameter.__init__(self, param_decls, type=type, **attrs)
TypeError: __init__() got an unexpected keyword argument 'hidden'
```

#### Option 2

```
$ sudo python3 -m pip uninstall -r requirements.txt
The directory '/home/rscm/.cache/pip/http' or its parent directory is not owned by the current user and the cache has been disabled. Please check the permissions and owner of that directory. If executing pip with sudo, you may want sudo's -H flag.
Not uninstalling setuptools at /usr/lib/python3/dist-packages, outside environment /usr
Uninstalling psutil-5.8.0:
  /usr/local/lib/python3.6/dist-packages/psutil-5.8.0.egg-info
  /usr/local/lib/python3.6/dist-packages/psutil/__init__.py
  /usr/local/lib/python3.6/dist-packages/psutil/__pycache__/__init__.cpython-36.pyc
  /usr/local/lib/python3.6/dist-packages/psutil/__pycache__/_common.cpython-36.pyc
  /usr/local/lib/python3.6/dist-packages/psutil/__pycache__/_compat.cpython-36.pyc
  /usr/local/lib/python3.6/dist-packages/psutil/__pycache__/_psaix.cpython-36.pyc
  /usr/local/lib/python3.6/dist-packages/psutil/__pycache__/_psbsd.cpython-36.pyc
  /usr/local/lib/python3.6/dist-packages/psutil/__pycache__/_pslinux.cpython-36.pyc
  /usr/local/lib/python3.6/dist-packages/psutil/__pycache__/_psosx.cpython-36.pyc
  /usr/local/lib/python3.6/dist-packages/psutil/__pycache__/_psposix.cpython-36.pyc
  /usr/local/lib/python3.6/dist-packages/psutil/__pycache__/_pssunos.cpython-36.pyc
  /usr/local/lib/python3.6/dist-packages/psutil/__pycache__/_pswindows.cpython-36.pyc
  /usr/local/lib/python3.6/dist-packages/psutil/_common.py
  /usr/local/lib/python3.6/dist-packages/psutil/_compat.py
[... 616 lines cut ...]

Total CPU usage: 0.3 %
Total system load: 0.45
Average temp. of all cores: 35.125 °C

Load optimal
setting turbo boost: off

-------------------------------------------------------------------------------

			"auto-cpufreq" refresh in: 4^C
Aborted!
```

---

### Conclusion

To sum it up

To install auto-cpufreq in elementary OS (at least)
 + `sudo apt remove -y python3-psutil python3-click python3-distro`
  * Probably just  `sudo apt remove -y python3-click`
 + `sudo ./auto-cpufreq-installer`
 + `sudo apt install -y elementary-desktop`

veerabag, 2021-07-17:
Extremely sorry guys as hell of a busy and was in a hectic situation. Thanks a million but I couldn't test the above steps as one of my coworker has changed the OS on the laptop which I tested Zorin.

gallowglass, 2021-08-01:
`auto-cpufreq` would not work for me as a snap or with the installer.  

Inspired by @rscm above, I removed `python3-click`, and whatever needed that, installed `auto-cpufreq`, then reinstalled `python3-click` and its dependents and things seem to be working.

FWIW, I am on Pop!_OS 18.04.

AdnanHodzic, 2021-08-09:
It seems in certain scenarios following packages will need to be removed and then reinstalled. It seems to be more of an environment/distro specific issue then auto-cpufreq itself. Hence I'll close this issue for the time being.

If any of you are still experiencing these problems, feel free to re-open the issue.

closed 2021-08-09 by AdnanHodzic