AdnanHodzic/auto-cpufreq #204: Warning: Changing CPU turbo is not supported. Skipping.

RiemaruKarurosu, 2021-05-31:
### Error output:
```text
Battery is: charging

Setting to use: "performance" governor

Total CPU usage: 6.2 %
Total system load: 1.33
Average temp. of all cores: 44.0 °C

Load optimal
setting turbo boost: off
Warning: Changing CPU turbo is not supported. Skipping.
```
---

### System information:
```text
-------------------------------------------------------------------------------

Linux distro: Manjaro Linux 21.0.5 Ornara
Linux kernel: 5.12.2-1-MANJARO
Processor: Intel(R) Core(TM) i5-8250U CPU @ 1.60GHz
Cores: 8
Architecture: x86_64
Driver: intel_pstate

------------------------------ Current CPU stats ------------------------------

CPU max frequency: 1600 MHz
CPU min frequency: 400 MHz

Core	Usage	Temperature	Frequency
CPU0:	  1.0%     42 °C     1800 MHz
CPU1:	  0.0%     41 °C     1800 MHz
CPU2:	  4.0%     44 °C     1800 MHz
CPU3:	  0.0%     42 °C     1800 MHz
CPU4:	  2.0%     42 °C     1800 MHz
CPU5:	  0.0%     41 °C     1800 MHz
CPU6:	  0.0%     44 °C     1600 MHz
CPU7:	  1.0%     42 °C     1800 MHz

auto-cpufreq version:

Python: 3.9.5
psutil package: 5.8.0
platform package: 1.0.8
click package: 7.1.2
distro package 1.5.0

Computer type: Notebook
Battery is: charging

auto-cpufreq system resource consumption:
cpu usage: 0.0 %
memory use: 0.14 %

Total CPU usage: 0.4 %
Total system load: 1.15
Average temp. of all cores: 42.25 °C 

Currently using: performance governor
Currently turbo boost is: off

-------------------------------------------------------------------------------

```
---

Why Turbo Boost is not supported, Am I missing something? or I need to overload my lap?

Intel® Core™ i5-8250U CPU @ 1.60GHz × 8 (Turbo Boost up to 3.40 GHz)

AdnanHodzic, 2021-06-01:
It gives you that message because neither `/sys/devices/system/cpu/intel_pstate/no_turbo` nor `/sys/devices/system/cpu/cpufreq/boost` are detected.

Can you give me output of `ls -al /sys/devices/system/cpu/`

RiemaruKarurosu, 2021-06-01:
```shell
total 0
drwxr-xr-x 18 root root    0 jun  1 14:14 .
drwxr-xr-x 10 root root    0 jun  1 14:14 ..
drwxr-xr-x  9 root root    0 jun  1 14:14 cpu0
drwxr-xr-x  9 root root    0 jun  1 14:14 cpu1
drwxr-xr-x  9 root root    0 jun  1 14:14 cpu2
drwxr-xr-x  9 root root    0 jun  1 14:14 cpu3
drwxr-xr-x  9 root root    0 jun  1 14:14 cpu4
drwxr-xr-x  9 root root    0 jun  1 14:14 cpu5
drwxr-xr-x  9 root root    0 jun  1 14:14 cpu6
drwxr-xr-x  9 root root    0 jun  1 14:14 cpu7
drwxr-xr-x 10 root root    0 jun  1 14:15 cpufreq
drwxr-xr-x  2 root root    0 jun  1 14:15 cpuidle
drwxr-xr-x  2 root root    0 jun  1 14:18 hotplug
drwxr-xr-x  2 root root    0 jun  1 14:15 intel_pstate
-r--r--r--  1 root root 4096 jun  1 14:18 isolated
-r--r--r--  1 root root 4096 jun  1 14:18 kernel_max
drwxr-xr-x  2 root root    0 jun  1 14:15 microcode
-r--r--r--  1 root root 4096 jun  1 14:18 modalias
-r--r--r--  1 root root 4096 jun  1 14:18 offline
-r--r--r--  1 root root 4096 jun  1 14:14 online
-r--r--r--  1 root root 4096 jun  1 14:18 possible
drwxr-xr-x  2 root root    0 jun  1 14:18 power
-r--r--r--  1 root root 4096 jun  1 14:18 present
drwxr-xr-x  2 root root    0 jun  1 14:18 smt
-rw-r--r--  1 root root 4096 jun  1 14:14 uevent
drwxr-xr-x  2 root root    0 jun  1 14:18 vulnerabilities
```

AdnanHodzic, 2021-06-01:
So `/sys/devices/system/cpu/intel_pstate` is there after all, can you please provide me with output of: 
`ls -al /sys/devices/system/cpu/intel_pstate`

RiemaruKarurosu, 2021-06-01:
```shell
total 0
drwxr-xr-x  2 root root    0 jun  1 17:32 .
drwxr-xr-x 18 root root    0 jun  1 17:31 ..
-rw-r--r--  1 root root 4096 jun  1 17:32 hwp_dynamic_boost
-rw-r--r--  1 root root 4096 jun  1 17:32 max_perf_pct
-rw-r--r--  1 root root 4096 jun  1 17:32 min_perf_pct
-rw-r--r--  1 root root 4096 jun  1 17:38 no_turbo
-r--r--r--  1 root root 4096 jun  1 17:38 num_pstates
-rw-r--r--  1 root root 4096 jun  1 17:38 status
-r--r--r--  1 root root 4096 jun  1 17:38 turbo_pct
```

Dejee98, 2021-06-04:
Same here my turbo-boost isn't working either.

ayushr01, 2021-06-11:
I'm facing the same issue on Fedora 34. I've tried both snap and the installer. This error doesn't happen immediately after installing the daemon. It only starts after the first reboot.

ayushr01, 2021-06-11:
I did some testing and I think this issue is due to the way my laptop's BIOS is configured. On boot up, when the temperature of the laptop is low turbo - boost is supported and my i5-8520u boots till 3.4Ghz:

[image]

However a few seconds after boosting, the temperature rises and the system disables turbo boost support:

[image]

When this happens autocpu-freq starts giving the error "Warning: Changing CPU turbo is not supported. Skipping.".

Eventually as the system cools down it is once again able to turbo and autocpu-freq is able to enable and disable turbo without a problem: 

[image]

FPSUsername, 2021-06-23:
At what temperature does your CPU run to have boost enabled? My XPS-15 9560 runs at ~50C when doing browsing and such on the low fan speeds (2500RPM). Not sure if it's like that in Windows as well.

AdnanHodzic, 2021-08-14:
Added [Troubleshooting section to README on how to change CPU management driver](https://github.com/AdnanHodzic/auto-cpufreq#troubleshooting).

Could you please give this a try as I believe it will resolve your problem. If it doesn't please feel free to re-open this issue.

closed 2021-08-14 by AdnanHodzic