AdnanHodzic/auto-cpufreq #216: [Documentation] Tell users to switch to integrated graphics

aindriu80, 2021-07-13:
I have been using auto-cpufreq on my XPS 15 9550 with Pop OS for a while but incredibly I had forgotten to switch from Nvidia discrete graphics to integrated graphics!  The difference is huge, maybe it's a bit too obvious but I have been using Windows mostly before.  Maybe it should be mentioned in the auto-cpufreq documentation.

[image]

AdnanHodzic, 2021-10-16:
With [v1.7.0 release](https://github.com/AdnanHodzic/auto-cpufreq/releases/tag/v1.7.0) it's now possible to manually define some of the settings instead of leaving everything to be picked up automatically. Could you try tweaking some of the [available options](https://github.com/AdnanHodzic/auto-cpufreq/#configuring-auto-cpufreq) and report back on how that worked for you? Thanks

AdnanHodzic, 2022-11-11:
> Maybe it should be mentioned in the auto-cpufreq documentation

Since I'm very limited with available time to do this myself, if you or anyone else wants to give it a try in implementing this functionality, please give it a try and [contribute to the project](https://github.com/AdnanHodzic/auto-cpufreq/#code-contribution) and you will be credited for your work as part of [future release](https://github.com/AdnanHodzic/auto-cpufreq/releases).

In meantime closing the issue due to inactivity.

closed 2022-11-11 by AdnanHodzic