AdnanHodzic/auto-cpufreq #239: `--install` does not work on AUR package

TrevCan, 2021-09-05:
Hey there, I tried running this and seems like an install script is missing? I understand that this works as a systemctl daemon so I enabled it and after testing with `--monitor` it appeared to be working. I'd suggest adding maybe a fallback to enable the systemctl command... Also, after disbaling the systemctl service running `auto-cpufreq --install` said already installed and running, which partly was incorrect. It was running, but it wasn't installed, because I disabled it. So I'd suggest also adding some checks to validate it on the system. Anyways, thanks for making this tool!

### Error output:
```text
Traceback (most recent call last):
  File "/usr/bin/auto-cpufreq", line 151, in <module>
    main()
  File "/usr/lib/python3.9/site-packages/click/core.py", line 1137, in __call__
    return self.main(*args, **kwargs)
  File "/usr/lib/python3.9/site-packages/click/core.py", line 1062, in main
    rv = self.invoke(ctx)
  File "/usr/lib/python3.9/site-packages/click/core.py", line 1404, in invoke
    return ctx.invoke(self.callback, **ctx.params)
  File "/usr/lib/python3.9/site-packages/click/core.py", line 763, in invoke
    return __callback(*args, **kwargs)
  File "/usr/bin/auto-cpufreq", line 131, in main
    deploy_daemon()
  File "/usr/lib/python3.9/site-packages/auto_cpufreq/core.py", line 300, in deploy_daemon
    shutil.copy(
  File "/usr/lib/python3.9/shutil.py", line 418, in copy
    copyfile(src, dst, follow_symlinks=follow_symlinks)
  File "/usr/lib/python3.9/shutil.py", line 264, in copyfile
    with open(src, 'rb') as fsrc, open(dst, 'wb') as fdst:
FileNotFoundError: [Errno 2] No such file or directory: '/usr/share/auto-cpufreq/scripts/auto-cpufreq-install.sh'
```
---

### System information:
```text
# auto-cpufreq --debug > .debug-auto-cpufreq
distro package 1.5.0

-------------------------------------------------------------------------------

Linux distro: Arch Linux rolling n/a
Linux kernel: 5.13.13-zen1-1-zen
Processor: Intel(R) Core(TM) i7-8565U CPU @ 1.80GHz
Cores: 8
Architecture: x86_64
Driver: intel_pstate

------------------------------ Current CPU stats ------------------------------

CPU max frequency: 800 MHz
CPU min frequency: 400 MHz

Core	Usage	Temperature	Frequency
CPU0:	  2.0%     44 °C     2000 MHz
CPU1:	  0.0%     44 °C     2000 MHz
CPU2:	  1.0%     44 °C     2000 MHz
CPU3:	  0.0%     44 °C     2000 MHz
CPU4:	  1.0%     44 °C     2000 MHz
CPU5:	  2.0%     44 °C      800 MHz
CPU6:	  1.0%     44 °C     2000 MHz
CPU7:	  3.0%     44 °C     2000 MHz

auto-cpufreq version:
Version         : 1.6.4.r228.5ca58e7-1

Python: 3.9.6
psutil package: 5.8.0
platform package: 1.0.8
click package: 8.0.1

Computer type: Notebook
Battery is: discharging

auto-cpufreq system resource consumption:
cpu usage: 0.0 %
memory use: 0.11 %

Total CPU usage: 2.1 %
Total system load: 0.87
Average temp. of all cores: 44.0 °C 

Currently using: powersave governor
Currently turbo boost is: off

-------------------------------------------------------------------------------
```
---

AdnanHodzic, 2021-09-13:
Thanks! 

Since I'm not someone who's involved in creation of AUR pqckage.

AdnanHodzic, 2021-09-13:
I

yochananmarqos, 2021-09-13:
@AdnanHodzic Maybe add a check for `/usr/share/auto-cpufreq/scripts/auto-cpufreq-install.sh`. If it doesn't exist, check if the service exists and is enabled / running.

AdnanHodzic, 2021-09-20:
Since as I said I don't use AUR and don't have a lot of time on my hands lately. 

If someone could create a PR with this change I'd gladly review it and of course tag you as a contributor for [that release](https://github.com/AdnanHodzic/auto-cpufreq/releases) if changes get merged,

TrevCan, 2021-09-20:
@AdnanHodzic ok, I'll look into it.

AdnanHodzic, 2022-02-20:
Closing the issue due to inactivity, if the problem persist please feel free to re-open the issue.

mizzunet, 2022-02-27:
I'm facing this issue
```
auto-cpufreq --install
sh: line 1: snap: command not found

--------------------- Deploying auto-cpufreq as a daemon ----------------------

* Turn off bluetooth on boot [skipping] (package providing bluetooth access is not present)

* Deploy auto-cpufreq install script
Traceback (most recent call last):
  File "/bin/auto-cpufreq", line 226, in <module>
    main()
  File "/usr/lib/python3.10/site-packages/click/core.py", line 1128, in __call__
    return self.main(*args, **kwargs)
  File "/usr/lib/python3.10/site-packages/click/core.py", line 1053, in main
    rv = self.invoke(ctx)
  File "/usr/lib/python3.10/site-packages/click/core.py", line 1395, in invoke
    return ctx.invoke(self.callback, **ctx.params)
  File "/usr/lib/python3.10/site-packages/click/core.py", line 754, in invoke
    return __callback(*args, **kwargs)
  File "/bin/auto-cpufreq", line 203, in main
    deploy_daemon()
  File "/usr/lib/python3.10/site-packages/auto_cpufreq/core.py", line 358, in deploy_daemon
    shutil.copy(SCRIPTS_DIR / "auto-cpufreq-install.sh", "/usr/bin/auto-cpufreq-install")
  File "/usr/lib/python3.10/shutil.py", line 417, in copy
    copyfile(src, dst, follow_symlinks=follow_symlinks)
  File "/usr/lib/python3.10/shutil.py", line 254, in copyfile
    with open(src, 'rb') as fsrc:
FileNotFoundError: [Errno 2] No such file or directory: '/usr/share/auto-cpufreq/scripts/auto-cpufreq-install.sh'
```

Installed from AUR on Artix Linux with `openrc`.
```
ls /usr/share/auto-cpufreq/scripts/
cpufreqctl.sh
```

mizzunet, 2022-03-01:
@AdnanHodzic Add these lines in PKGBUILD
```
    install -Dm755 scripts/auto-cpufreq-install.sh -t "$pkgdir/usr/share/${pkgname%-git}/scripts"
    install -Dm755 scripts/auto-cpufreq-openrc -t "$pkgdir/usr/share/${pkgname%-git}/scripts"
    install -Dm755 scripts/auto-cpufreq-remove.sh -t "$pkgdir/usr/share/${pkgname%-git}/scripts"
    install -Dm755 scripts/auto-cpufreq-runit -t "$pkgdir/usr/share/${pkgname%-git}/scripts"
    install -Dm755 scripts/auto-cpufreq-venv-wrapper -t "$pkgdir/usr/share/${pkgname%-git}/scripts"
    install -Dm755 scripts/auto-cpufreq.service -t "$pkgdir/usr/share/${pkgname%-git}/scripts"
    install -Dm755 scripts/cpufreqctl.sh -t "$pkgdir/usr/share/${pkgname%-git}/scripts"
    install -Dm755 scripts/snapdaemon.sh -t "$pkgdir/usr/share/${pkgname%-git}/scripts"
```

Here's the full file https://0x0.st/oPzq.txt

mizzunet, 2022-03-02:
I shall try to change it then

AdnanHodzic, 2022-06-10:
Seems to be related to #388, let's continue the discussion there.

closed 2022-06-10 by AdnanHodzic