AdnanHodzic/auto-cpufreq #247: Whisker-menu not closing properly power save governor.

Sokoloft, 2021-09-28:
Hello.

I've had an issue for a few where my whisker menu would not close when I'd press my super key. I'm using Manjaro XFCE. It seems to be when my laptop is in the power save governor. When it's charging I do not have this issue. As well as when the service for auto-cpufreq is stopped. There is no problem.

Possibly this is a problem with xcape or Manjaro? Or a bug with auto-cpufreq? I'm unsure.

I have a post [here ](https://forum.manjaro.org/t/whisker-menu-not-closing-correctly/84558)on the Manjaro forums of me slowly figuring this out. Any help would be greatly appreciated. Thanks.

AdnanHodzic, 2021-10-01:
I really don't see how this would have any correlation to auto-cpufreq, hence I don't think it's an auto-cpufreq bug.

Would be interested to find out what was the problem in the end.

Sokoloft, 2021-10-01:
Well it isn't completely the fault of auto-cpufreq, but more the fact that it sets my laptop to the power save governor. When in that mode, my whisker menu doesn't close properly. I have to double press the binding. By default, my laptop is in the schedutil mode. 

So I suppose I'll try another post on Manjaro forums or possibly post an issue on the xcape github. I'll post back here once I have an answer. Thank you for the package. It's been working well up until I discovered this issue.

Sokoloft, 2021-10-02:
Hello. If you go to my Manjaro post someone commented saying they reported the issue to the whisker menu gitlab and it seems their is a fix. Just need to wait for it to be pushed to the Manjaro repo. Cheers. 

https://gitlab.xfce.org/panel-plugins/xfce4-whiskermenu-plugin/-/issues/61

closed 2021-10-02 by Sokoloft