AdnanHodzic/auto-cpufreq #251: Not scaling CPU frequency

Hidan0, 2021-10-04:
### Error output:
```text
CPU frequency it's not scaling properly.
I installed the tool from the AUR. I've tried changing intel driver from GRUB but it still says `intel_pstate` and not scaling. 
```
---

### System information:
```text
-------------------------------------------------------------------------------

Linux distro: ArcoLinux rolling n/a
Linux kernel: 5.14.8-arch1-1
Processor: 11th Gen Intel(R) Core(TM) i5-1135G7 @ 2.40GHz
Cores: 8
Architecture: x86_64
Driver: intel_pstate

------------------------------ Current CPU stats ------------------------------

CPU max frequency: 1400 MHz
CPU min frequency: 400 MHz

Core	Usage	Temperature	Frequency
CPU0:	  8.9%     42 °C     1400 MHz
CPU1:	100.0%     44 °C     1400 MHz
CPU2:	  7.0%     44 °C     1400 MHz
CPU3:	  7.1%     44 °C     1360 MHz
CPU4:	  9.0%     42 °C     1400 MHz
CPU5:	  4.0%     44 °C     1400 MHz
CPU6:	 10.8%     44 °C     1303 MHz
CPU7:	  6.0%     44 °C     1400 MHz

auto-cpufreq version:
fatal: not a git repository (or any of the parent directories): .git
CalledProcessError(128, ['git', 'describe', '--always'])

Python: 3.9.7
psutil package: 5.8.0
platform package: 1.0.8
click package: 8.0.1
/bin/sh: line 1: pip3: command not found
distro package

Computer type: Convertible
Battery is: charging

auto-cpufreq system resource consumption:
cpu usage: 0.0 %
memory use: 0.21 %

Total CPU usage: 19.1 %
Total system load: 1.95
Average temp. of all cores: 43.5 °C 

Currently using: performance governor
Currently turbo boost is: off

-------------------------------------------------------------------------------
```
---

AdnanHodzic, 2021-10-16:
With [v1.7.0 release](https://github.com/AdnanHodzic/auto-cpufreq/releases/tag/v1.7.0) it's now possible to manually define some of the settings instead of leaving everything to be picked up automatically. Could you try tweaking some of the [available options](https://github.com/AdnanHodzic/auto-cpufreq/#configuring-auto-cpufreq) and report back on how that worked for you? Thanks

AdnanHodzic, 2021-10-17:
> Hi there, after toggling some settings in the UEFI the driver now is 'acpi-cpufreq'. '--stats' is showing that my CPU cores in "powersave" are capped at 400MHz (min) except for one core (~1000MHz) but cpupower-gui is showing that all cores are running at >~1000MHz. Am I doing something wrong?

Could you please upgrade to [v1.7.1 ](https://github.com/AdnanHodzic/auto-cpufreq/releases)which was just released and report back if you still have the same problem? As one of the issues that was related to powersave mode. 

If settings aren't picked up right automatically, I'd encourage you to [set these settings](https://github.com/AdnanHodzic/auto-cpufreq/#configuring-auto-cpufreq) to your liking.

AdnanHodzic, 2022-02-20:
Just [released v1.9.2](https://github.com/AdnanHodzic/auto-cpufreq/releases) which adds ability to change power governor used by default with auto-cpufreq from `balanced` to `performance` [using power_helper.py script](https://github.com/AdnanHodzic/auto-cpufreq/#1-power_helperpy-script) which could also resolve this problem.

In meantime closing the issue due to inactivity, if the problem persist please feel free to re-open the issue.

closed 2022-02-20 by AdnanHodzic