AdnanHodzic/auto-cpufreq #275: Systemd not auto starting on boot even after the service is enabled

KnightKrusty, 2021-11-11:
### Error output:
```text
auto-cpufreq.service - auto-cpufreq - Automatic CPU speed & power optimizer f>
     Loaded: loaded (/etc/systemd/system/auto-cpufreq.service; enabled; vendor >
     Active: inactive (dead)

```
### it works but i have to start manually

```
auto-cpufreq.service - auto-cpufreq - Automatic CPU speed & power optimizer f>
     Loaded: loaded (/etc/systemd/system/auto-cpufreq.service; enabled; vendor >
     Active: active (running) since Thu 2021-11-11 14:10:26 IST; 3s ago
   Main PID: 6465 (auto-cpufreq)
      Tasks: 1 (limit: 8689)
     Memory: 15.6M
        CPU: 422ms
     CGroup: /system.slice/auto-cpufreq.service
             └─6465 /usr/bin/python3 /usr/local/bin/auto-cpufreq --daemon

```

---

### System information:
```text
Linux distro: Fedora Linux 35 
Linux kernel: 5.14.16-301.fc35.x86_64
Processor: AMD Ryzen 5 4600H with Radeon Graphics
Cores: 12
Architecture: x86_64
Driver: acpi-cpufreq

------------------------------ Current CPU stats ------------------------------

CPU max frequency: 3000 MHz
CPU min frequency: 1400 MHz

Core	Usage	Temperature	Frequency
CPU0:	  8.2%     38 °C     1400 MHz
CPU1:	  3.0%     38 °C     3000 MHz
CPU2:	  7.1%     38 °C     1412 MHz
CPU3:	  5.0%     38 °C     3000 MHz
CPU4:	  6.1%     38 °C     1400 MHz
CPU5:	  4.1%     38 °C     1400 MHz
CPU6:	  2.0%     38 °C     1400 MHz
CPU7:	  4.9%     38 °C     1400 MHz
CPU8:	  3.0%     38 °C     1400 MHz
CPU9:	  1.0%     38 °C     1400 MHz
CPU10:	  3.0%     38 °C     1400 MHz
CPU11:	  4.0%     38 °C     1400 MHz

auto-cpufreq version:
fatal: not a git repository (or any of the parent directories): .git
CalledProcessError(128, ['git', 'describe', '--always'])

Python: 3.10.0
psutil package: 5.8.0
platform package: 1.0.8
click package: 8.0.1
distro package 1.6.0

Computer type: Notebook
Battery is: discharging

auto-cpufreq system resource consumption:
cpu usage: 0.0 %
memory use: 0.29 %

Total CPU usage: 4.8 %
Total system load: 1.53
Average temp. of all cores: 37.5 °C 

Currently using: schedutil governor
Currently turbo boost is: on

```
---

titulebolide, 2021-11-12:
Same issue, also on Fedora 35
Apparently it is an issue with pkg_resources being very old and deprecated
https://github.com/mtkennerly/dunamai/issues/28

titulebolide, 2021-11-12:
Edit : Upgrading autocpufreq just by installing it again fixed the error

KnightKrusty, 2021-11-12:
@titulebolide 

are you sure? i reinstalled it but that didnt fixed it for me...

have you restarted your device and then checked? and
manually starting the service make it active for few minutes and then kill it self. so make sure to check the service after 5 to 10 mintues later.

titulebolide, 2021-11-12:
Actually when I reinstalled it I trid to have it removed (when it asked me what to do), and the error popped again. Then I launched again the installator and have it to install and it worked this time

KnightKrusty, 2021-11-12:
@titulebolide  ohh ok, I will try and update here if it fixes the issue..

AdnanHodzic, 2021-11-14:
I've noticed this as well on other distros, in case like above, best course of action is to uninstall auto-cpufreq and all Python dependencies mentioned in [auto-cpufreq-installer](https://github.com/AdnanHodzic/auto-cpufreq/blob/master/auto-cpufreq-installer) and then re-install auto-cpufreq which will also re-install necessary dependencies.

KnightKrusty, 2021-11-19:
@titulebolide @AdnanHodzic 

Was not able to update regarding the reinstall and if it fixed my issue.. so in my case reinstalled didnt fixed the issue but read few similar post in  issue i found out that service is being killed by gnome power profile daemon..

> Yeah, stopping & masking the gnome power profile daemon should fix your problem.
> 
> ```
> systemctl stop power-profiles-daemon
> ```
> 
> Followed by
> 
> ```
> systemctl mask power-profiles-daemon
> ```
> 
> Then either reboot or start auto-cpufreq
> 
> ```
> systemctl start auto-cpufreq
> ```

here is there link to that post.. https://github.com/AdnanHodzic/auto-cpufreq/issues/277#issuecomment-967797641

closed 2021-11-19 by KnightKrusty

AdnanHodzic, 2021-11-21:
I've made some changes which will resolve this problem automatically without any workarounds, for more info please refer to #277 and let's continue the discussion there.