AdnanHodzic/auto-cpufreq #276: issues on pop_os

srulikuk, 2021-11-12:
I have some issues on pop_os

1. 
```
user@pop-os:~$ sudo auto-cpufreq --monitor
...
...
...
  File "/snap/auto-cpufreq/77/lib/python3.8/site-packages/distro.py", line 1064, in _os_release_info
    with open(self.os_release_file) as release_file:
PermissionError: [Errno 13] Permission denied: '/etc/os-release'
```
Turns out in pop_os the /etc/os-release file is symlinked to another file, as a workaround i removed the symlink and copied the file in its place, you might want to add a check to see if the file is symlinked maybe follow to main file or something.
 
2. this is most likely not the right place to ask but you might know so giving it a shot, pop_os does NOT use grub, instead uses systemd-boot (never used it before) would you know how to change from `intel_pstate` to `acpi-cpufreq` ?

3. for some reason "auto-cpufrq" is not listed in the pop_os store, i installed the snap from the command line however tab wouldnt auto complete the name, and typing in the terminal "auto-cpu" wont auto-complete with tab.

Thanks

AdnanHodzic, 2021-11-14:
> I have some issues on pop_os
>
>user@pop-os:~$ sudo auto-cpufreq --monitor
...
...
...
  File "/snap/auto-cpufreq/77/lib/python3.8/site-packages/distro.py", line 1064, in _os_release_info
    with open(self.os_release_file) as release_file:
PermissionError: [Errno 13] Permission denied: '/etc/os-release'
Turns out in pop_os the /etc/os-release file is symlinked to another file, as a workaround i removed the symlink and copied the file in its place, you might want to add a check to see if the file is symlinked maybe follow to main file or something.

I've seen this error and it's solution before mentioned in few other issues, like: #240 

>this is most likely not the right place to ask but you might know so giving it a shot, pop_os does NOT use grub, instead uses systemd-boot (never used it before) would you know how to change from intel_pstate to acpi-cpufreq ?
>
>for some reason "auto-cpufrq" is not listed in the pop_os store, i installed the snap from the command line however tab wouldnt auto complete the name, and typing in the terminal "auto-cpu" wont auto-complete with tab.

Sorry but both of these seem to be related strictly to your environment, hence I can't help you with that. I would ask someone from Pop OS side.

andrej-zirko, 2021-11-25:
This is a popos issue, where the /etc/os-release is symlinked to the /etc/pop-os/os-release. The solution is to create a hard-link instead of the symlink. It's working for me. Hope it helps you as well. 

`
sudo mv /etc/os-release /etc/os-release-backup
sudo ln /etc/pop-os/os-release /etc/os-release
`

WUZICANGJIE, 2021-12-13:
> This is a popos issue, where the /etc/os-release is symlinked to the /etc/pop-os/os-release. The solution is to create a hard-link instead of the symlink. It's working for me. Hope it helps you as well.
> 
> `sudo mv /etc/os-release /etc/os-release-backup sudo ln /etc/pop-os/os-release /etc/os-release`

It works! Thanks!

AdnanHodzic, 2021-12-16:
Since this seems to be a recurring issue for Pop OS, if someone can fix this once and for all and [make a contribution](https://github.com/AdnanHodzic/auto-cpufreq/#code-contribution) as part of [auto-cpufreq-installer](https://github.com/AdnanHodzic/auto-cpufreq/blob/master/auto-cpufreq-installer) or add it as part of [core functionality](https://github.com/AdnanHodzic/auto-cpufreq/blob/master/auto_cpufreq/core.py) (so it also works if installed using a snap package) that would be great.

Thanks

AdnanHodzic, 2021-12-19:
Thanks to @tyleraharrison contribution, this problem is now permanently fixed for Pop OS!

Changes are available on `master` branch and are part of [1.8.2 release](https://github.com/AdnanHodzic/auto-cpufreq/releases).

closed 2021-12-19 by AdnanHodzic