AdnanHodzic/auto-cpufreq #289: Thermald overlap

mhalano, 2021-11-22:
Hi! Great piece of code! I just have a question which could become part of documentation: what is the overlap with thermald? It's clear the auto-cpufreq is a replacement for TLP, but I couldn't find any information about the integration (or not) with thermald.

AdnanHodzic, 2021-11-22:
auto-cpufreq works great with thermald and I'd even recommended to have both auto-cpufreq and thermald running at the same time (that's how I have it).

Will close this issue, if you still have any questions and/or comments, feel free to re-open it.

closed 2021-11-22 by AdnanHodzic

mhalano, 2021-11-22:
@AdnanHodzic Can you document it on README? I think more people could have this same question in the future.

AdnanHodzic, 2021-11-22:
> @AdnanHodzic Can you document it on README? I think more people could have this same question in the future.

Sure, as this question does keep popping up, added as part of ["Why do I need auto-cpufreq"](https://github.com/AdnanHodzic/auto-cpufreq/blob/master/README.md#why-do-i-need-auto-cpufreq).