AdnanHodzic/auto-cpufreq #290: Tuned overlap

mhalano, 2021-11-22:
I did the same question for thermald, now I ask for tuned. There is any overlap between two solutions? auto-cpufreq was projected to replace TLP, but there is any problem to use with tuned?

AdnanHodzic, 2021-11-23:
No overlap, they should work fine in tandem.

closed 2021-12-05 by AdnanHodzic

dutrajulio, 2024-06-19:
I tested "tuned" today and now it appears to be incompatible with "auto-cpufreq".

Maybe something in the new release, v2.23.0, of "tuned".

Anyone can confirm this too?

yabo-boye, 2025-03-18:
I have also not been succesful in making those two programs overlap!

AdnanHodzic, 2025-03-18:
> I have also not been succesful in making those two programs overlap!

@yabo-boye Why are the exact issues are experiencing when using both?