AdnanHodzic/auto-cpufreq #296: auto-cpufreq stops after reboot

InsaneFarhan, 2021-11-30:
### Error output:
```text
Paste here error output
```
---

### System information:
```text
Paste here output of the auto-cpufreq --debug
```
---

InsaneFarhan, 2021-11-30:
i need to use "sudo auto-cpufreq --install" command after every reboot

InsaneFarhan, 2021-11-30:
if i  install it using snap 
it only stays in powersaver mode even when charger plugged in
yeah i do not need to install it again

bobslept, 2021-11-30:
Maybe this is issue is related to issue #277.
You can try the gnome-power branch and see if that fix your problem.

InsaneFarhan, 2021-12-01:
> Maybe this issue is related to issue #277. You can try the gnome-power branch and see if that fix your problem.

i ran these commands

InsaneFarhan, 2021-12-01:
> Maybe this issue is related to issue #277. You can try the gnome-power branch and see if that fix your problem.

thanks i ran these commands
https://github.com/AdnanHodzic/auto-cpufreq/issues/277#issuecomment-968246100
it solved the problem

i wanna ask one more think
can these command cause any problem
i am asking because these commands disable power-profiles-daemon

bobslept, 2021-12-01:
> > Maybe this issue is related to issue #277. You can try the gnome-power branch and see if that fix your problem.
> 
> thanks i ran these commands [#277 (comment)](https://github.com/AdnanHodzic/auto-cpufreq/issues/277#issuecomment-968246100) it solved the problem
> 
> i wanna ask one more think can these command cause any problem i am asking because these commands disable power-profiles-daemon

I see you manualy ran the commands. The gnome-power branch would do this for you automaticly. Glad to hear that it solved your problem. 

It should not bring any problems. power-profiles-daemon is doing power management, which is in conflict with the power management that auto-cpufreq will do. So auto-cpufreq will take over, but in a slightly different way. I'm running for about a week now with the power-profiles-daemon disabled and havent seen any errors or problems so far.

AdnanHodzic, 2021-12-04:
Hi everyone, could you please test final changes before I merge them with `master` and make them part of 1.8.0 release.

Latest changes are available on `gnome-power-v2` branch, which you test by running:

```
git fetch
git checkout gnome-power-v2
sudo ./auto-cpufreq-installer --install
```

For those using Snap package changes are available as part of beta channel which can be installed by running:

```
sudo snap install auto-cpufreq --beta
```

Thanks!

AdnanHodzic, 2021-12-05:
Functionality which addresses this problem has now been released as part of [1.8.0 release](https://github.com/AdnanHodzic/auto-cpufreq/releases).

closed 2021-12-05 by AdnanHodzic